KenT2/tboplayer #20: Interacting with youtube-dl causes interface to get locked until youtube-dl finishes

heniotierra, 2015-09-04:
youtube-dl may take some seconds to return the result and until then the GUI is locked,
I guess a way to avoid this is to use subprocess instead of pexpect. Or migrate TBOPlayer to python3 and use pexpect asynchronously.

Do you have an opinion, @KenT2 ?

KenT2, 2015-09-05:
@heniotierra could you contact me here, https://pipresents.wordpress.com/contact/, I would like  to discuss TBOPlayer development by email.

For omxplayer, pexpect spawns a subprocess, sends it commands and listens for and filters replies using '.expect' 'pexpect.expect' is blocking so the listening is done in a python thread  in _get_position and signals set in this thread are polled in the main program in  play_state_machine. 

play_state_machine uses Tkinter's after method which sleeps in away which allows gui events to be received while waiting.

youtube_dl looks similar to omxplayer in that you kick it off and then want to wait until the video loads. I would interface it in the same way as omxplayer using a second instance of pexpect and a second thread to monitor its output. You probably need a much enhanced play_state_machine which listens for signals from both threads.

Hope this helps

heniotierra, 2015-09-05:
Oh yes, this is so much helpful! Thanks for your input, it's much appreciated. :+1:

KenT2, 2015-09-05:
Glad to be of help.

The youtube-dl suggested that you need to completely download the video before starting omxplayer. I wonder if you can minimise the delay by starting omxplayer while the video is downloading. There would need to be a certain amount downloaded to allow for pauses in transmission etc.

heniotierra, 2015-09-05:
Actually, the arguments I used for youtube-dl extracts the media link from the page and then this link is what is passed to omxplayer. This way omxplayer plays the video stream directly. When adding Youtube playlists, it downloads the playlist as JSON and then this is processed and the direct links are extracted from the JSON "manually". :)

heniotierra, 2015-09-06:
Ok, I've implemented this, and it is working great. I implemented the Ytdl class (youtube-dl interface) with all of it's methods as static methods. I thought this might be better for readability, but I'm not quite sure about that. Of course changing this doesn't require any effort at all. Maybe it doesn't really matter.

heniotierra, 2015-09-06:
Also, I though of using some indent fixer to make the the code more organized. But then again, I don't know if that's necessary.

heniotierra, 2015-09-07:
I left the indentation alone for now.

closed 2015-09-07 by heniotierra