KenT2/tboplayer #40: Installing TBOPlayer on Raspbian

uksa, 2015-09-20:
I had a few issues with installing the TBOPlayer, this was on a fresh install however without the following it won't work:

1. sudo apt-get install python-pip
2. pip install pexpect
3. pip install ptyprocess
4. apt-get install python-gobject-2

heniotierra, 2015-09-20:
Thanks for the input. :) 

I had not tested it on a fresh Raspbian install. 
The installer should install pexpect, but maybe there's some some catch related to ptyptocess.
gobject was recently added as a requirement for DBus connectivity, but I thoguht it would be pre-installed.

That will be fixed soon. I'll send the patch along with a new feature for TBOPlayer.

uksa, 2015-09-20:
I'm sure pexpect was fine, however I think this was the full list. I've also tested in a 3.5" screen and it seems to be working ok so :)

Close the issue when ever you like...

heniotierra, 2015-09-21:
This should be fixed by https://github.com/KenT2/tboplayer/pull/41 now.

closed 2015-09-21 by heniotierra

somesmart, 2016-12-09:
I just installed this on Raspbian and had to manually install pexpect and python-magic in order to get it to work. Let me know if you need more information!

heniotierra, 2016-12-09:
Weird. Which version of Raspbian did you install? Thank you.

somesmart, 2016-12-09:
I've had it for a while so I'm still on Rasbian 7 (wheezy)

heniotierra, 2016-12-09:
What these two packages have in common is that thay both are installed with pip. So my guess is that your problem is related to pip. Did you have to install pip too? Did you use pip normally to install the packages?

somesmart, 2016-12-10:
No, pip was already installed. I was just ran the scripts from the manual installation instructions:

`pip install --user pexpect ptyprocess python-magic`