KenT2/tboplayer #45: sudo pip install should not be used

Lartza, 2015-11-22:
In the installer and instructions, sudo pip install is used which is dangerous at worst and unstable at best. Packages from the OS repositories should be used instead, or another way.

heniotierra, 2015-11-24:
Yes, that's because instaling pexpect through apt-get got me problems when trying to load ptyprocess, while pip installs makes it work with no problem. And installing pexpect/ptyprocess from source also gave me the same problem as apt-get, so I had no choice.

Why do you say pip is dangerous though? I get its packages are not so stable, but I'm unaware of any security issues aside of that.

Lartza, 2015-11-24:
Running sudo pip runs setup.py as root, and that can have any code inside it.

The stability problem is less about the pip packages being stable, but your system being less stable when you go around your package manager and touch system files with pip. Worst case is obviously updating already existing packages but installing new ones ain't much better.

pip install --user or a virtualenv should be used instead
There is even a long standing issue on pip about this pypa/pip/#1668

heniotierra, 2015-11-24:
Thanks, I will investigate this later.

heniotierra, 2015-12-08:
This has been corrected: the README.md and setup.sh both have pip install --user.

closed 2015-12-08 by heniotierra