KenT2/tboplayer #46: No module named gtk.gdk

shadecrawler, 2015-12-11:
Traceback (most recent call last):
  File "/home/pi/tboplayer/tboplayer.py", line 392, in <module>
    from gtk.gdk import ( screen_width, screen_height )
ImportError: No module named gtk.gdk

Hi, I'm getting the above error after installing it either manually or the "first way". Is there an easy solution to this? 
Unfortunately I'm a total linux noob, so sorry if the solution is overly simple.

heniotierra, 2015-12-11:
Are you using Raspbian? Which version?

Try `sudo apt-get install python-gtk2`  in the terminal and see if that helps.

shadecrawler, 2015-12-13:
Awesome, that did the trick for me. Thanks!
I'm using the Wheezy version. It only says  May 2015...

heniotierra, 2015-12-13:
I'm glad that you're good. I've added python-gtk2 to the installation in setup.sh and README.md, so, closing the issue.

closed 2015-12-13 by heniotierra