KenT2/tboplayer #47: Pause at the end

jmggs, 2016-01-20:
Hi, 
Is possible to cue a video like start paused at first frame?

And at the end of the video freeze at the last frame? 

I want to use the TBOPlayer as a playout server and that is very usefull. 

Thanks

KenT2, 2016-01-20:
Is possible to cue a video like start paused at first frame?
No, but it would not be too difficult to add it.

And at the end of the video freeze at the last frame? 
No, but Pi Presents version 1.3 (gapless) will.

You might find PI Presents a better fit for your application.

heniotierra, 2016-01-20:
Hi,

Yes, it'd be possible add an option for this in the Options panel. For pausing at the start, just start the video and then immediately force it to pause. For pausing at the end of the video you can calculate what would be the time for the last frame, based on video length and frames per second, and then pausing at the right time.

However, being pessimistic, I'm not sure if that would be very precise. I mean, it should generally work well, but ensuring it would be paused at that exact time may be problematic, since depending on system load a few frames could play before the video would be paused, and if passes that point, then it'd have lost the chance.

I can try to do it! But it'd have to wait, since ATM I'm really busy... You're welcome to do it yourself!

heniotierra, 2016-01-20:
Oh, and there's Pi Presents! ;)

KenT2, 2016-01-20:
In Pi Presents Version 1.3 (gapless) I do pause at start but not in the way that the OP wants it. I had to modify JBaiters code a little to get it to work reliably. Look at omxdriver.py

For pause at end no calculation is necessary. You can run omxplayer with the -i option to obtain the duration, I then subtract some mS. and pause omxplayer at that time. The time is got from the timecode returned by omxplayer with the -s option. You miss a couple of frames at the end.

heniotierra, 2016-01-20:
I'm aware of the -i option (TBOPlayer now uses it too), but I didn't know about -s. And TBOPlayer's Jbaiter has been heavily modified to fit the new features, but I guess this could be put this in.

jmggs, 2016-01-20:
I'm not a programmer and i have no ideia how to do it :(

But pause the end at could be a option like repeat, simple, playlist. 

For the start if is not possible to pause at first frame, can be at frame 2, 3 without any problem. At the end the same. It's not a problem. Because i use tboplayer to playout to a blackmagic atem video switcher and the videos have a few seconds more at the beginng and at the end. 

In pro video and broadcast we use playout software that do that. But i'm using TBOplayer for a small project for a school. It will be great if this features can be implemented. There are many users like me. 

Thanks

jmggs, 2016-01-21:
Do you guys think is possible to implement this?

heniotierra, 2016-01-23:
I'll try to do this this sunday, but can give no guarantee. If not possible, probably next week. It should not be too much work, so I guess it would take a few hours of work.

jmggs, 2016-01-24:
Thanks you are the best!

heniotierra, 2016-01-25:
@jmggs I have implemented this. It's a "Start/End paused " option in the Options menu, but no option to either  start OR end paused. Is that what you want? It's working, but I didn't upload it yet.

jmggs, 2016-01-25:
Hi,
 I think is that. So i press play one time starts paused, if press again is starts and end paused? If is that its ok ;)

Thanks -------- Original message --------

heniotierra, 2016-01-26:
Ok then. Uploaded it now. If you find a problem, please report.

heniotierra, 2016-01-26:
Oh, you should close the thread if you think the issue is solved. ;) Cheers.

jmggs, 2016-01-26:
Hi, It's working!!! 
Just one thing... is possible that the pause at beginning start at frame 1 (00:00:01) instated of frame 0 (00:00:00)? 
Because at most of the videos the frame 0 is black and frame 1 have video information.

Thank you so much! 
You are the best! 

P.S. Is in your plans to add a counter in reverse - remaining time?

heniotierra, 2016-01-26:
Well, it would be possible, but other users would have different needs. So, I'm sorry, but If you have such specific needs, you'd have to edit the source yourself.

jmggs, 2016-01-26:
It's fair..Thanks for the add on ;)

closed 2016-01-26 by jmggs

heniotierra, 2016-01-26:
You have the option to edit the videos too.

heniotierra, 2016-01-26:
About the time counter in reverse, I tried that, but it was not possible because I could not find a way to make the background of the time element to be transparent, which screwed the looks.

jmggs, 2016-01-26:
I was talking about counter in reverse at the gui. I'm using the TBOPlayer via VNC to display 1. 
The display 0 is the output of the Raspberry HDMI out and i have a option of omxplayer to --no-osd 

So its a clean ouput. 

I have disabled all desktop icons and i have a black background and the start menu is hidden. 

The count down is usefull for the director give times to the talent and the concrete time to switch the video input. 

The freeze at the end is very usefull for the video not go black on air.  The same at beginning for preview the video and also for acidental switch without going black. But most of the videos at frame 0 are black. I will try if i can change to frame instead of frame 0. 

Thanks for you work. 
There are any version of the gui over web?

heniotierra, 2016-01-26:
Yes, me too, was talking about the same thing. problem is, I didnt find a way to make this look good, since Tkinter does nor provide the necessary functionality, which is transparency.

If you are using tboplayer in a vnc desktop, you may find the option "forbid windowed mode" useful.

There is a web gui https://github.com/brainfoolong/omxwebgui

heniotierra, 2016-03-06:
@jmggs Hey, I just thought of a more obvious, painless way to show current time/total time. Before, I was trying to put the time display OVER the progress bar, but realized it would not be possible, because transparency would be necessary for that, which is a nono for Tkinter.

KenT2, 2016-03-06:
For a gui over the web I have been using the gui toolkit RemI in Pi Presents https://github.com/dddomodossola/remi
It would not be too difficult to implement the TBOPlayer gui using this.