KenT2/tboplayer #48: [Raspberry Pi 3] Videos appear displaced?

heniotierra, 2016-04-09:
Seems like videos are shown displaced by ~50px in the Raspberry Pi 3. I just noticed this, and ATM I have no idea of what might be the cause. Some investigation is required.

heniotierra, 2016-04-09:
It would be welcome if someone could test if this also happens in Raspbian Jessie with the Raspberry Pi 1.

KenT2, 2016-04-10:
Sounds like overscan problems. Run PP with the -d command line option.It will print out the screen size that it has obtained from Raspbian.

heniotierra, 2016-04-10:
Hi Ken, the player now uses the functions ```screen_width``` and ```screen_height``` to get the screen resolution, which, yes, is used to place the video in the right place. I guess I must have made a mistake somewhere.

KenT2, 2016-04-10:
Whoops, I answered the question thinking it was a Pi Presents issue. My suggestion of overscan not being set correctly by the OP is still relevant. It  would make the result of your correct corrections incorrect.

heniotierra, 2016-04-10:
Then, please, enlighten me, sir! I do not know what overscan is nor how that affects TBOplayer. May HDMI cause that kind of problem?

koeshi, 2016-04-10:
You can disable Overscan by following Menu>Preferences>Raspberry Pi Configuration and then setting Overscan to Disabled.

heniotierra, 2016-04-11:
Oooh! That's it! It worked like a charm! Thank you @koeshi !

heniotierra, 2016-04-11:
I'm gonna add this to the Troubleshooting section of the README.md file.

closed 2016-04-11 by heniotierra