KenT2/tboplayer #49: Select playlist with key bindings - TBOplayer

twarr, 2016-06-04:
Hi!

First of all: thank you for a great piece of software! I'm a newbie at raspberry pi all the way from Norway, and I habe a question about the TBOplayer.

I have a project where I need to select a specific playlist easily with just a click of a button, with a board like this: http://img.dxcdn.com/productimages/sku_428281_1.jpg

When I press the green button it selects playlist "rock", when I click the blue button it selects playlist "pop" and so on.

Is this something that is possible to fix in a easy way? :)

Thanks!

Best regards,

Aslak

heniotierra, 2016-06-04:
Hi twarr!

What you want is doable with TBOPlayer with some minor modifications. The idea would be to load up TBOPlayer passing a playlist as argument. However, to play another playlist you'd have to close the current TBOPlayer instance and then start a new one passing another playlist as argument. If that sounds good fo you, I can do it when I get some free time.

There are a few other options of software you could find interesting for that though, for example:
omxcontrol
https://github.com/rikkertkoppes/omxcontrol
omxdirector
https://github.com/alepez/omxdirector
or a simple bash script
http://www.cenolan.com/2013/03/looping-video-playlist-omxplayer-raspberry-pi/

You're welcome!

heniotierra, 2016-06-04:
Forgot to mention that TBOPlayer is already capable of adding files automatically upon starting, except that it does not work for playlists. To do it, call the program like this:

```python tboplayer/tboplayer.py '/path/to/a/file.mp4' '/path/to/another/file.mp4' '/path/to/yet/another/file.mp4'```

heniotierra, 2016-06-29:
implemented possibility to load a playlist from command line, so closing this

closed 2016-06-29 by heniotierra