KenT2/tboplayer #50: Missing uninstall docs

Schabernack, 2016-06-13:
Please provide documentation how to uninstall  tboplayer.

heniotierra, 2016-06-13:
To uninstall TBOPlayer, execute in the terminal:
```
rm -Rf ~/tboplayer && rm -f ~bin/tboplayer
```

If you want to remove the dependencies, you must be aware that doing so may make other programs that use the same stuff to stop working. To remove all dependencies, do:
```
sudo apt-get remove python-gobject-2 python-gtk2 python-requests libav-tools ffmpeg
sudo rm -f /usr/local/bin/youtube-dl
yes | pip uninstall pexpect ptyprocess
```

I should add this to the docs.

Schabernack, 2016-06-14:
Thank you for the quick reply.

heniotierra, 2016-06-27:
Now with commit https://github.com/KenT2/tboplayer/commit/9371a3113920e0e1fa2c267fe26f95e858b3eee9 tboplayer can be easily removed with the command ```~/tboplayer/setup.sh uninstall```.

closed 2016-06-27 by heniotierra