KenT2/tboplayer #54: I can't get this to work

tpitman, 2016-08-09:
I followed the installer instructions.

I made sure there is not another omxplayer instance running.

I tried playing the same file using the command line given in trouble shooting and it plays full screen just fine.

I use the Add button or menu to add the same file and nothing happen then or if I hit play.

I then tried playing the file again using the command line and it works fine.

heniotierra, 2016-08-09:
Does this happen to all of your files, or just to some specific files,
or file extensions?

tpitman, 2016-08-09:
I only have the one file for testing at this point. It is the bunny video that comes stock on a brand new Raspbian installation.

It is in the /opt/vc/src/hello_pi/hello_video folder called video.h264 or something like that.

heniotierra, 2016-08-09:
If the extension.nsion of the file is h264, then the reason is the file extesion. Rename your file to video.mp4 and try again. 

I could add this extension to the List of valid extensions, but doing a mimetype check instead of extension check is probably a good way of helping avoiding this kind of problem.

tpitman, 2016-08-09:
Ah, I was assuming it would just be passing the file along to the command line omxplayer because it plays the file just fine. I didn't realize that your wrapper cared what the file was named.

heniotierra, 2016-08-09:
It did not matter how it was named , but this was added as a way of restricting what kind if files were acceptable (igoring non-media files). Tkinter does not support extension restriction when selecting a folder, not that I know. So, that's done with a file extension check. Another way to do this (probably a better way) is with a mime-type check.

heniotierra, 2016-08-29:
Fixed this with #55 ... by adding the h264 (and another ones too) extension to the list of allowed extensions, which is not perfect.

After some research and testing, I found python does not have a good way of guessing the mime type of a file by default. That means we will need an external ~~MAGIC~~ library just for that. Might be worth it though. I will probably get to that soon.

heniotierra, 2016-08-29:
Using ~!!MAGIC!!~ now! Switched from extension checking to mime-type checking. The problem with unrecognized extensions should be solved now...

closed 2016-08-29 by heniotierra