KenT2/tboplayer #56: Unable to play video, error on index

syariffudinzo, 2016-10-03:
I've installed as per guide in the documentation. at first on fresh install the video can be played as expected. after changing the configuration and set it to "loop: repeat" at the configuration, it seems that I've trace an issue as below in console:

`Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1535, in __call__
    return self.func(*args)
  File "/opt/tboplayer/tboplayer.py", line 610, in play_track
    self.play()
  File "/opt/tboplayer/tboplayer.py", line 507, in play
    self.start_omx(self.playlist.selected_track_location)
  File "/opt/tboplayer/tboplayer.py", line 876, in start_omx
    self.omx = OMXPlayer(track, args=opts, start_playback=True)
  File "/opt/tboplayer/tboplayer.py", line 115, in __init__
    self.make_dict()
  File "/opt/tboplayer/tboplayer.py", line 163, in make_dict
    if index != 0: self.failed_play_signal = True
UnboundLocalError: local variable 'index' referenced before assignment
`

Steps: 
1. install per guide from documentation
2. run TBO player through shell/icon
3. select open dir
4. select media and play
5. stop media and go to configuration
6. set media loop as repeat
7. check the force window thing.
8. play it again.
9. exit
10. start again by selecting folder 
11. try play.
12 observe
[image]

heniotierra, 2016-10-03:
I will investigate this when I find some free time. Thank you.

heniotierra, 2016-10-05:
Hi Syariffudin Sapri, are you sure the file that`s to be played works?
Maybe it`s corrupt? Is it a specific file? You see, I have already
made a fix to avoid the error, however this error seems to indicate
that the file failed to be played by omxplayer. If you try to open
this file with omxplayer directly on the command line, does it work?

2016-10-03 13:06 GMT-03:00, Hênio Tierra <heniotster@gmail.com>:
> I will investigate this when I find some free time. Thank you.
>

heniotierra, 2016-10-10:
@syariffudinzo Can you please test whether the change I made solves the problem? To update your version of tboplayer to the version in the temporary branch, do:

        sudo wget https://raw.githubusercontent.com/KenT2/tboplayer/fix_issue_56/tboplayer.py -O /opt/tboplayer/tboplayer.py

And then execute it normally.
As I said in the previous comment, this version should avoid the error, however the error seems to indicate that the file failed to be played by omxplayer. It would be more helpful if you could provide the details I requested above.

If you have no time it`s OK. Later on I will get to the tests myself.

heniotierra, 2016-10-13:
The fix is in the master branch now. (not the same version as the one above.

closed 2016-10-13 by heniotierra