KenT2/tboplayer #57: Unsuccsessful installation

gugica, 2016-10-26:
When attempting to install via auto installation....
cd ~ && wget https://github.com/KenT2/tboplayer/tarball/master -O - | tar xz &&
cd KenT2-tboplayer-* && chmod +x setup.sh && ./setup.sh

...I get the following message during the installation. Even though I get at the end of the installation process a message which is reporting successful installation, command tboplayer is not working after that.

[image]

I would appreciate your Help.

heniotierra, 2016-10-26:
Hi,

Which OS (and version) are you using?
Also, closing the terminal after installation and then opening a new
terminal, tboplayer command still does not work?

gugica, 2016-10-26:
I am using Raspbian Jessie with XFCE desktop.
No luck with opening a new terminal window though :(

heniotierra, 2016-10-26:
Okay... can you find Tboplayer in your apps menu, under multimedia or
related? If not, then for now you can run tboplayer with the command
'''python /opt/tboplayer/tboplayer.py'''. I think this problem must be
because of incompatibility with xfce. It's fixable, but might require
specific folder definition for lxde.

gugica, 2016-10-26:
I figure it out! Omxplayer wil not launch if I type "omxplayer" command in to terminal but that's OK. Running "python /opt/tboplayer/tboplayer.py" Gave me some kind of error that Dbus is missing on my system.
`sudo apt-get install python-dbus` Fixed the problem and now is working perfectly :)

heniotierra, 2016-10-26:
I'm glad you've figured it out. 

It's weird that you can't run omxplayer through the terminal, though. Tboplayer requires omxplayer to be properly installed and working. I hope you are being able to play stuff properly with Tboplayer.

I should add python-dbus package to the installer. Previously it would come pre-installed with stock Raspbian. I'll look into the problem with XFCE and other desktop enviroments, and see if it's worth fixing.

Thank you! :)

heniotierra, 2016-10-28:
@gugica Your issue with XFCE has been fixed. To try it, just follow the setup instructions again. It will make tboplayer command work for you and also make file associations as it must so that you can open files with double-click. Works for any desktop manager.

closed 2016-10-28 by heniotierra