KenT2/tboplayer #59: Alsa Mixer

Octopus3535, 2016-12-21:
This is probably wrong as I am learning about linux and raspbiean, but im having trouble with the alsa mixer. There isn't an option in the TBOplayer and I have been trying to make it the default. My headset is a usb headset, and i has it's own alsa mixer. Is there a way to have alsa come through the TBOplayer? I'm sorry if this is a problem on my end, it probably is, but i haven't found anything that works.

Thanks,
Octopus3535

heniotierra, 2016-12-21:
I'm sorry, but omxplayer, which is used by tboplayer, does not provide
support for ALSA. If you want to use ALSA, you'll have to resort to another
player. You can try vlc, for example. Just don't expect to have hardware
accelaration for your videos.

>
> This is probably wrong as I am learning about linux and raspbiean, but im
having trouble with the alsa mixer. There isn't an option in the TBOplayer
and I have been trying to make it the default. My headset is a usb headset,
and i has it's own alsa mixer. Is there a way to have alsa come through the
TBOplayer? I'm sorry if this is a problem on my end, it probably is, but i
haven't found anything that works.

heniotierra, 2016-12-21:
On a second thought, I think VLC probably can't access the audio jack, and that may be true for all players that don't use omxplayer, unless you change the default configuration. 
You can check here to see how to change the said config:
https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=40872

KenT2, 2016-12-21:
The version of omxplayer in the sept 2016 raspbian supports alsa.

heniotierra, 2016-12-22:
In this case, @Octopus3535 , you can just use '-o alsa' in the TBOPlayer's options panel.

eysispeisi, 2017-01-03:
i'm working on some changes, they include adding an an alsa checkbox in options.

heniotierra, 2017-01-04:
Hey, the resizing thing is cool! 
However, about the ALSA support part, I don't think a radio button will work for the case of the OP. A Checkbox should be better, allowing this pattern in the cli:  '''-o local:alsa'''

KenT2, 2017-01-04:
There is ongoing work which will change the omxplayer interface in this area:

[https://github.com/popcornmix/omxplayer/issues/498](url)

heniotierra, 2017-01-04:
I have merged the PR by eysispeisi. Since it's a simple implementation, it most probably will be supported after the changes to omxplayer.

closed 2017-01-04 by heniotierra

eysispeisi, 2017-01-04:
My version of omxplayer already supports it... is there a new change coming or am i missing something?

my version -->
omxplayer - Commandline multimedia player for the Raspberry Pi
        Build date: Fri, 23 Sep 2016 00:56:49 +0000
        Version   : dfea8c9 [master]
        Repository: https://github.com/popcornmix/omxplayer.git

heniotierra, 2017-01-04:
Yes, it is supported. I meant it will STILL be supported after the changes mentioned in the linked thread.