KenT2/tboplayer #64: Problem with ExceptionCatcher

heniotierra, 2017-01-08:
@eysispeisi I don't think the player should exit upon catching errors related to DBUS connection, because the user should still be able to control omxplayer, except not using the DBUS interfaces' functions.

heniotierra, 2017-01-09:
Hi @eysispeisi , I'm sorry for not making this more clear. Making the player exit upon encountering a Tkinter breaking exception is a good thing. However, DBus exceptions don't break the interface. You could specify an exception handling for DBus exceptions with ```except dbus.DBusException, e:``` .

eysispeisi, 2017-01-09:
yah, your right, my mistake, it was pretty clear what you said. i made this change while going on only a few hours of sleep, which i should really stop doing.
lets just close this port, i have other fixes i want to do as well before i do another port.

heniotierra, 2017-01-10:
I found another bug in add_url method that occurred when the clipboard was empty, and also fixed this here. There might still be other kind of exceptions that should not make TBOPlayer to quit. We shall see.

closed 2017-01-10 by heniotierra