KenT2/tboplayer #66: Drag and drop support

eysispeisi, 2017-01-08:
Been combing the internet for a drag and drop solution for tkinter, simply isn't there in this version of tkinter that is currently compiled for any python implementation.

Other UI bindings do have support for DnD, like gtk.
I did a messy hack, combining the tkinter tboplayer with a gtk DnD window.
https://github.com/eysispeisi/tboplayer/tree/dnd

So much about this seems dubious though, first it's adding gtk to tboplayer, dont remember if it's a default package or if it needs to be installed.

Anyways this change has a long way to go, but you can test it out if you want, any comments would be appriciated.

heniotierra, 2017-01-10:
Only now I had the chance to test this out. It's working good, but hum, aestetically it doesn't fit. I thought of implementing this drag n drop feature, but realised my best option was to convert the whole program to GTK, so... yeah, I dropped the idea.

eysispeisi, 2017-01-13:
Yah, converting to GTK requires a complete overhaul of everything. with this hack we can atleast implement it until a DnD option is added to tkinter.

heniotierra, 2017-01-14:
We don't know if this will ever go into TKinter. I mean, ut should, but would tke so long... his implementation of yours has a few problems. Ideally, the drop button should be the list itself, but the way it is it's so out of place. Other problem is that TBOplayer has two seperate windows in the applications bar, and it you focus out of TBOPlayer and then click the gtk window, all you'll see is the gtk window; Also, as is, this button should not be resizable.

About another way to implement this for TKinter, please, check the answer here: http://stackoverflow.com/questions/14267900/python-drag-and-drop-explorer-files-to-tkinter-entry-widget

eysispeisi, 2017-01-14:
TkDND looks a lot better, did a quick mocup of it in https://github.com/eysispeisi/tboplayer/tree/tkdnd
it only works on files at the moment, not on urls from browser

eysispeisi, 2017-01-14:
note, with this a few setup commands are needed...

sudo apt-get install tk-dev tcl-dev
git clone https://github.com/petasis/tkdnd ; cd tkdnd
./configure
make
sudo make install

heniotierra, 2017-01-14:
Is it really needed to compile it? I think tkdnd is available in apt repositories.

heniotierra, 2017-01-14:
Also, it's better to leave TKDnD class inside tboplayer.py file..

heniotierra, 2017-01-15:
merged

closed 2017-01-15 by heniotierra