KenT2/tboplayer #73: youtube-dl not copied in Ubuntu Mate for Raspberry Pi

aaryadev, 2017-01-27:
Hi when installed in Ubuntu Mate for Raspberry Pi (16.04)  youtube-dl is not copied to /usr/local/bin.
Have to manually copy it from /usr/bin.

when launch is from /opt/tboplayer/tboplayer.py its shows youtube-dl is not found in /usr/local/bin, so manually copied it . it worked out of box.

In Pixel no such issue found

heniotierra, 2017-01-27:
Hi @aaryadev , tell me, youtube-dl was already installed when you installed tboplayer, am I correct? Seems to me like you had already installed youube-dl from ubuntu repository, using apt-get?

aaryadev, 2017-01-27:
No it's fresh install. I installed it again. still same , but not happening with PIXEL and UBUNTU 64 desktop

aaryadev, 2017-01-27:
error on Ubuntu Mate 16.4 Raspi

`Traceback (most recent call last):
  File "/opt/tboplayer/tboplayer.py", line 2784, in <module>
    bplayer = TBOPlayer()
  File "/opt/tboplayer/tboplayer.py", line 1219, in __init__
    self.ytdl.check_for_update(ytdl_updated)
  File "/opt/tboplayer/tboplayer.py", line 419, in check_for_update
    current_version_hash = sha256(open(self._YTLOCATION, 'rb').read()).hexdigest()
IOError: [Errno 2] No such file or directory: '/usr/local/bin/youtube-dl'`

so I manually copied from /usr/bin

heniotierra, 2017-01-27:
Then you are saying that the installer installs youube-dl in /usr/bin instead of /usr/local/bin ? That would be very weird. Apt-get installs a very old version of tboplayer in thr /usr/bin directory, that's why I thought it was the case...

Anyway, I think I know how to deal with this problem... can you do a test after I push a possible fix?

aaryadev, 2017-01-27:
Yes sure , Thanks a lot , I am installing 12 Raspi so i can test it any time.

heniotierra, 2017-01-27:
@aaryadev , please, try this setup script:
https://raw.githubusercontent.com/KenT2/tboplayer/fix_73/setup.sh
You can substitute this content (ctrl-a, ctrl-c, ctrl-v) with the content of your setup.sh script in /opt/tboplayer/, and then just run it. My approach here is check if youtube-dl is already installed, and if it is, it will be removed for reinstalling. 
If that does not solve the issue, my guess is that this is some problem with wget, althoigh I have no basis for this assumption.  approach will be to try using curl instead of wget. Do you think curl comes pre-installed in your distro?

heniotierra, 2017-01-27:
Another thing that must be done is to just avoid using youtube-dl if it cannot be found in the path configured in the options.

aaryadev, 2017-01-27:
CUrl is preinstalled. yes on fresh install i checked youtube-dl and and youtube-dl-gui is pre-installed so...

closed 2017-01-27 by aaryadev

heniotierra, 2017-01-27:
Right, good! Now, I still have to avoid trying to use youtube-dl if it's not in the path configured in the options, and then tell the user youtube-dl could not be found.

heniotierra, 2017-01-27:
That's done. If youtube-dl is not in the configured path, tboplayer informs the user and tells him to fix his config.

heniotierra, 2017-01-27:
Setup has not been touched.