KenT2/tboplayer #84: usb DVB-t tuner

pejko, 2017-04-03:
Hi this is not and issue more like a question. Is it possible to load/play tv chanels usinf tboplayer and an usb dvbt tunner? i managed to find the drivers for tunner and test it with vlc so drivers are fine, just need to know how to hook it together with tboplayer.
Ty for your time and this great GUI :)

heniotierra, 2017-04-03:
Hi @pejko , I believe you should be able to either open the device (inside /dev) directly as a normal track, or you could get a url which then you'd paste into tboplayer's url dialog. If omxplayer can play it, then tboplayer must be capable of that too. Or did you try those things already without success?

pejko, 2017-04-03:
wow thank you for such fast reply , unfortunately i have to reinstall raspbian  :D but will write back asap .

heniotierra, 2017-04-09:
I guess you've had your problem solved, so I'll close this issue. If you need more help, you can reopen it.

closed 2017-04-09 by heniotierra