KenT2/tboplayer #88: no Drag and Drop module

beckpeter, 2017-04-09:
I just did a fresh install of Raspbian on the pi (2 B+).  Just installed this package using the automatic installer and it gives me the following error:
_tkinter.TclError: can't find package tkdnd

closed 2017-04-09 by beckpeter

heniotierra, 2017-04-09:
So... what happened? The installer should install tkdnd automatically, unless there is some problem with it. Like, idk, the download being corrupt. Although I think apt-get must have some way to avoid this...

beckpeter, 2017-04-09:
Disclaimer: I am fairly new with the raspberry pi, so it more than likely is something on my end that I don't know about.

I can't find tkdnd using:  sudo apt-get install tkdnd
it gives me an error of 'unable to locate package tkdnd'. 

I have tried 2 different mirrors on it (kind of out of necessity because the mirror director isn't working for me).   

And I wouldn't know how to manually install tkdnd

I just ended up commenting out the dnd command in the python code, because its not too important to me.

heniotierra, 2017-04-09:
Ok, thanks for your answer. Now, if you want to use the drag n drop feature, use the commands below to download, build and install tkdnd package from source code:

sudo apt-get install build-essential gcc tcl-dev tk-dev
wget https://github.com/petasis/tkdnd/tarball/master -O - | tar xz &&
cd petasis-tkdnd-*
./configure
make
sudo make install