KenT2/tboplayer #92: Access basic controls via command line

agustinscaz, 2017-04-15:
It would be great to have to control the GUI with commands like:
```
python /opt/tboplayer/tboplayer.py pause
python /opt/tboplayer/tboplayer.py play
python /opt/tboplayer/tboplayer.py next
python /opt/tboplayer/tboplayer.py prev
python /opt/tboplayer/tboplayer.py vol-up
python /opt/tboplayer/tboplayer.py vol-down
```

Could you make this possible?

Thanks!

heniotierra, 2017-04-15:
Well, yes, it can be done by exposing the respective methods to the DBus interface. However, to control these DBus methods via CLI, it would be best to connect directly to the DBus layer with a shell script instead of calling tboplayer as you are proposing, because by calling tboplayer this way, the Pi would have to load the whole tboplayer code to memory, which takes time.
 
May I ask what exactly you are trying to achieve? Are you writing your own script to controll tboplayer or what?

agustinscaz, 2017-04-15:
Im trying to control the tboplayer with the voice with this project (https://github.com/alexylem/jarvis).
What I have done is adding items to playlist like this:
subprocess.Popen('python /opt/tboplayer/tboplayer.py ' + csvFile, shell=True)

So I say, _jarvis, play rock_, and I load a list of N *.mp3 files.
Then if I say _jarvis, play pop_ the tboplayer adds more songs to the list, but what I want is to skip songs, pause, etc...

heniotierra, 2017-04-15:
Ok, in this case, I just need to expose the methods to the DBus intarface as I said. Your part would be to connect to TBOPlayer via DBus and  call the method exposed. 

By the way, currently you can open files using the DBus interface too. The way you are doing it files uses DBus connection, but takes longer for the same reason I stated above, that is, having to load the whole TBOPlayer code to memory.

This is the code to connect to tboplayer`s dbus interface and open files:

```
TBOPLAYER_DBUS_OBJECT = "org.tboplayer.TBOPlayer"
TBOPLAYER_DBUS_PATH = "/org/tboplayer/TBOPlayer"
TBOPLAYER_DBUS_INTERFACE = "org.tboplayer.TBOPlayer"

import dbus

# connect tho TBOPlayer's DBus interface
bus = dbus.SessionBus()
bus_object = bus.get_object(TBOPLAYER_DBUS_OBJECT, TBOPLAYER_DBUS_PATH, introspect = False)
dbusif_tboplayer = dbus.Interface(bus_object, TBOPLAYER_DBUS_INTERFACE)

# call openFiles method passing a array containing file paths
dbusif_tboplayer.openFiles()
```

heniotierra, 2017-04-15:
Oh, you must have guessed it, but to use the DBus interface, tboplayer has to be running already. If it`s not, then you`ll have to start tboplayer first with the command you posted.

heniotierra, 2017-04-16:
@agustinscaz There, it's done.

The DBus interface methods are:

openFiles - expects [String] array as argument
openPlaylist - expects String as argument
openUrl - expects String as argument
play - expects track index (integer) as argument (defaults to 0)
pause
stop
next
previous
volumnDown
volumnUp

closed 2017-04-16 by heniotierra

agustinscaz, 2017-04-16:
I can't make it work, and I don't understand what's the problem, can you help me?

File: test.py
`
TBOPLAYER_DBUS_OBJECT = "org.tboplayer.TBOPlayer"
TBOPLAYER_DBUS_PATH = "/org/tboplayer/TBOPlayer"
TBOPLAYER_DBUS_INTERFACE = "org.tboplayer.TBOPlayer"

import dbus

# connect tho TBOPlayer's DBus interface
bus = dbus.SessionBus()
bus_object = bus.get_object(TBOPLAYER_DBUS_OBJECT, TBOPLAYER_DBUS_PATH, introspect = False)
dbusif_tboplayer = dbus.Interface(bus_object, TBOPLAYER_DBUS_INTERFACE)

# call openFiles method passing a array containing file paths
#dbusif_tboplayer.openFiles(['/media/USB/Music/Otros/Bso - Star Wars Imperial March.mp3','/media/USB/Music/Otros/Bso - Rocky Song - Gonna Fly Now.mp3'])

# play - pause songs
dbusif_tboplayer.playPause()
`

File: tboplayer.py (the only change is in this class, I add def playPause... )
`
class TBOPlayerDBusInterface (Object):
    tboplayer_instance = None

    def __init__(self, tboplayer_instance):
        self.tboplayer_instance = tboplayer_instance
        dbus_loop = DBusGMainLoop(set_as_default=True)
        bus_name = dbus.service.BusName(TBOPLAYER_DBUS_OBJECT, bus = dbus.SessionBus(mainloop = dbus_loop))
        Object.__init__(self, bus_name, TBOPLAYER_DBUS_PATH)

    @dbus.service.method(TBOPLAYER_DBUS_INTERFACE, in_signature = 'as')
    def openFiles(self, files):
        self.tboplayer_instance._add_files(files)

    @dbus.service.method(TBOPLAYER_DBUS_INTERFACE, in_signature = 'as')
    def playPause(self):
        self.tboplayer_instance.toggle_pause()
`

I try with and withoug the `@dbus.service.method(TBOPLAYER_DBUS_INTERFACE, in_signature = 'as')` with always the same error:
`
Traceback (most recent call last):
  File "/opt/tboplayer/lolo.py", line 16, in <module>
    dbusif_tboplayer.playPause()
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/dbus/service.py", line 654, in _message_cb
    (candidate_method, parent_method) = _method_lookup(self, method_name, interface_name)
  File "/usr/lib/python2.7/dist-packages/dbus/service.py", line 246, in _method_lookup
    raise UnknownMethodException('%s is not a valid method of interface %s' % (method_name, dbus_interface))
UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: playPause is not a valid method of interface org.tboplayer.TBOPlayer
`

Thanks a lot!

heniotierra, 2017-04-16:
But you modified the code, right? Without your modifications, are you able to control the interface? 

The problem in your modification is that you are defining the method to receive an array of strings as argument.

However, although there is a pause method, the play method pauses the currently playing track if the player`s playing something. That should probably be changed to force playing of specified track.

heniotierra, 2017-04-16:
@agustinscaz 
I have changed the behaviour if the play method. In the current version, if no track is playing, it plays the track, or it pauses the track if that same track is currently playing, or if a different track is playing, it stops the track and starts the track specified.

Also, added fullscreen method to the interface.