KenT2/tboplayer #109: First start "Play" - "Pause" - Position seek

loriszigliani, 2017-08-08:
When I open TBO Player for the first time after raspberry start, the position seek, menu button "Play" not change status and rest Play, and position seek not appear, but if I play any movie and click next, the gui show position seek and button "play" change status to "Pause". 
After that, if I don't close TBOPlayer, the GUI works perfectly.
When i start Raspberry I'll do this step.

heniotierra, 2017-08-09:
Hi @KenT2,

Thanks for bringing this up, but I'm not sure I was able to completely understand your problem. You are saying that after starting the Raspberry Pi, you start tboplayer and the play button won't change to "Pause" after you press play to start playing a track and the same occurs with the progress bar, and only after hitting the "next" button the GUI will work as expected, is that it? 

if you can provide some log info regarding this problem, it will make easier to solve the bug. To log stuff, go to the options panel, select the "debug" option, save, and then restart tboplayer. (And in case the problem you mention only occurs the fisrt time you start tboplayer, right after starting the OS, then you should have to restart the OS too of course.)

Also, make sure you are using the latest version of tboplayer, which is dated "21 Jul 2017".

heniotierra, 2017-11-13:
Closing this, since there was no feedback, and I was unable to reproduce the said problems in the current version.

closed 2017-11-13 by heniotierra