KenT2/tboplayer #111: "Translators" requested

heniotierra, 2017-11-12:
TBOPlayer has just gained an internationalization system for itself, but is still lacking the translation files. Google translator or some other service can do the hard work for you, you just have the review and fix the produced translation. If you have interest, the instructions below should be enough to get the job done.

1) First, you'll need to have poedit installed, a software for making gettext translation files.

`sudo apt-get install poedit`

2) Then with poedit, open the .pot file at the root of the application repository.

3) At this point you might want to use google or some other service for automatically translating the texts so that you then just have to check and correct the bad translations produced, or you could choose to make the translation by hand, in which case you should proceed to step 4.

To use the help of an automatic translator, export the pot file as HTML. Go to File -> Export as HTML. Load the generated HTML in your preferred translator set the target language, and voila. Just make sure the translation won't have bad mistakes.

4) Go to File -> New, choose the language you will make the translation to, then click a text and in the corresponding area, fill with the translation (or copy-paste from another source, ofc), and save your work when finished.

5) Make a commit and pull request or send the translation files you just created ( .po and .mo files ) to heniotster at g m a i l dot com

fabianmendes, 2021-04-16:
How can I contribute to this fr Spanish?

heniotierra, 2021-04-19:
@fabianmendes If you think you can improve the current spanish translation, you just have to follow the instructions in the orginal post.