KenT2/tboplayer #116: cannot start tboplayer

fonggarry, 2018-01-04:
Hi, I am very new to this. I have installed the tboplayer by: 

cd ~ && wget https://github.com/KenT2/tboplayer/tarball/master -O - | tar xz &&
    cd KenT2-tboplayer-* && chmod +x setup.sh && ./setup.sh

After the installation, when I goto the icon and hit tboplayer - nothing happens.
I use Terminal window and manually type in tboplayer, I have: 

/usb/lib/python2.7/dist-packages/gtk-2.0/gtk/__init__.py:57:Gtkwarning - couldnot open display
warning.warn (str(e),gtk.warning)

Traceback (most recent call last); File "opt/tboplayer/tboplayer.py" line 2682, in <module> import requests

import Error: no module named requests

Am I missing something?

Thanks for your help in advance

heniotierra, 2018-01-04:
Hi,

The error says you don't have python-requests installed. The installer should have taken care of that, but something must have gone wrong.

Make sure that you have up-to-date data about packages with:
        sudo apt-get update

Then install python-requests:
        sudo apt-get install python-requests

Did you try the manual installation (instructions in the README file)?

fonggarry, 2018-01-04:
Thank you for your help. After installed the python-requests, when i invoke tboplayer, I got this error: 

/usr/lib/python2.7/dist-packages/gtk-2.0/gtk/__init__.py:57: GtkWarning: could not open display
  warnings.warn(str(e), _gtk.Warning)
Traceback (most recent call last):
  File "/opt/tboplayer/tboplayer.py", line 3173, in <module>
    bplayer = TBOPlayer()
  File "/opt/tboplayer/tboplayer.py", line 1124, in __init__
    self.root = tk.Tk()
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1823, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: no display name and no $DISPLAY environment variable

Perhaps I should un-install all tboplayer including the dependencies and re-install it again manually? 

Thanks.

heniotierra, 2018-01-04:
What linux distribution are you using?

Now the error says the screen can't be accessed. I have seen that happen when one's not using a desktop manager (a GUI environment). Uses to happen if you're trying to start TBOPlayer from an ssh terminal, or without having started a display server. If you are indeed using ssh, use VNC instead, that will solve your problem.

I think uninstalling would probably make no difference, but you can do a manual installation over the automatic installation, no problem.

fonggarry, 2018-01-04:
Hi, you are right. The ssh does not work. What I did was invoke from my pi. The GUI turned on (it took awhile but is ok). I can select and play video. However, the video play but can't see my screen because it cannot sync. I think that it is because I am using a 7 inch touch screen. And Somehow, tboplayer was displaying other resolution.  I have not problem invoke my mp4 from omxplayer and it sync with my screen. Just tboplayer cannot. Maybe I should edit the config file somewhere? I am very glad that tboplayer came up. That mean I have installed right. 

Thank you again for your help.

heniotierra, 2018-01-04:
F11 to go fullscreen mode? I hope that helps.

No problem.

closed 2018-01-05 by heniotierra