KenT2/tboplayer #137: TBOplayer runs as root, but it does not run as a regular user.

seem-less, 2018-06-03:
Hello,

When I run TBOplayer as root it works fine. When I try to run with a regular user I get this error:

```
student@raspberrypi:/opt/tboplayer $ python tboplayer.py
Traceback (most recent call last):
  File "tboplayer.py", line 60, in <module>
    from ytdl import *
  File "/opt/tboplayer/ytdl.py", line 2, in <module>
    import pexpect
ImportError: No module named pexpect

```

pexpect is located in the following directory:

`/home/pi/.local/lib/python2.7/site-packages/pexpect`

the user "student" home directory is located here:

`/media/home/student/`

Any idea why the pexpect module can't be found?  Thanks

heniotierra, 2018-06-03:
Hi,

Hum... Yes, your home user path is very likely the problem. I mean, when I wrote this code, I didn't expect the user the have a path different than /home/username/ ... 

I will see if I can fix this.

Thanks for your report,

heniotierra, 2018-06-03:
UPDATED ANSWER ABOVE

KenT2, 2018-06-03:
you can now do sudo apt-get install python-pexpect.

heniotierra, 2018-06-03:
Good, that should be enough to fix the installer. :)

heniotierra, 2018-06-04:
Fix must be fixed now.

closed 2018-06-04 by heniotierra