KenT2/tboplayer #138: tboplayer not launching; cannot import name from_file

vliegher, 2018-06-05:
Tboplayer is not launching. Neither from the desktop or the menu. When I try to launch it from a terminal window it stops as well. The last line is: Line 74 in <module> from magic import from_file Import Error : cannot import name from_file.
Is there a solution?

heniotierra, 2018-06-05:
Yes, just install pip if you don't have it, and the install python-magic lib.

sudo apt-get install python-pip
pip install --user python-magic

I have removed the installation of python-magic through pip, thinking that the package installed by apt-get, which is homonymous would work too, but they must be different libs. That's the reason for the error.

Thank you for pointing that out,

heniotierra, 2018-06-05:
Confirmed, python-magic from apt-get repositories is a different version than the one used by tboplayer.

commit e0432381 (2018-06-05): fix #138 : cannot import name from_file
closed 2018-06-05 by heniotierra