KenT2/tboplayer #142: save playlist: file not containing file paths

ttqf, 2018-08-10:
hi, have updated to the latest version and it seems that playlist saving is not working properly. the csv file does not contain file paths, only files names.
it is a clean update, removed all files and config.
[image]

heniotierra, 2018-08-10:
It will check this when I get some free time. Thank you.

ttqf, 2018-08-10:
problem seems to be in this function: `def save_list(self)`
line:
`ofile.write ('"' + (item if 'http' in item else self.playlist.selected_track()[PlayList.LOCATION_BACKUP]) + '","' + self.playlist.selected_track()[PlayList.TITLE]+'"\n')`
should be:
`ofile.write ('"' + (item if 'http' in item else self.playlist.selected_track()[PlayList.LOCATION]) + '","' + self.playlist.selected_track()[PlayList.TITLE]+'"\n')`

with this change it seems to be working. let me know if that should be the fix.

heniotierra, 2018-08-10:
Actually, I believe the problem is that there a 'not' operator missing before 'http'. Try this:

ofile.write ('"' + (item if not 'http' in item else self.playlist.selected_track()[PlayList.LOCATION_BACKUP]) + '","' + self.playlist.selected_track()[PlayList.TITLE]+'"\n')

If you make this change and it works as expected, you can send us a patch and I will merge it. Follow the contributions guideline in case you want to contribute. https://github.com/KenT2/tboplayer/blob/master/CONTRIBUTING.md

commit 0d32b5b9 (2018-08-11): Merge pull request #143 from KenT2/issue142

fix #142 : added missing not operator
commit ffa3421b (2018-08-11): fix #142 : added missing not operator
closed 2018-08-11 by ttqf