KenT2/tboplayer #144: URL playback not working

ttqf, 2018-08-11:
hi, tried to play some YouTube. the search function works. however the playback doesn't happen. it's stuck at the [wait] stage. from the debug log:
`
2018-08-11 22:48:18,241 - /opt/tboplayer/debugging.py - ERROR - Traceback (most recent call last):
  File "/opt/tboplayer/tboplayer.py", line 495, in ytdl_state_machine
    self.treat_ytdl_result(url, process[1])
  File "/opt/tboplayer/tboplayer.py", line 525, in treat_ytdl_result
    self.treat_video_data(url, result)
  File "/opt/tboplayer/tboplayer.py", line 545, in treat_video_data
    self.playlist.replace(track[0],[media_url, data['title'],track[1][PlayList.LOCATION_BACKUP]])
IndexError: tuple index out of range
`
with youtube-dl I can download and afterwards play the video with omxplayer. so it seems something is broken in the code.
tried other URLs and the same issue. could you look into this? thanks!

heniotierra, 2018-08-11:
I solved this already. Uploading...

heniotierra, 2018-08-11:
Thank you too.

commit 5fa4d131 (2018-08-11): fix #144
commit a9c9ca8a (2018-08-11): Merge pull request #145 from KenT2/issue144

fix #144
closed 2018-08-11 by heniotierra