KenT2/tboplayer #150: Error while starting to play

Maksim-Bartoshyk, 2018-10-25:
Just installed tboplayer, played a bit with it, then used ./setup uninstall and installed once again.
Have no luck to launch any url from youtube.
If trying to fetch url with ytdl an play directly with omx it works fine.

018-10-25 23:05:33,688 - /opt/tboplayer/debugging.pyc - ERROR - Traceback (most recent call last):
  File "/opt/tboplayer/tboplayer.py", line 1074, in set_progress_bar
    self.progress_bar_step_rate = self.omx.timenf['duration']/self.progress_bar_total_steps
KeyError: 'duration'

2018-10-25 23:05:34,044 - /opt/tboplayer/debugging.pyc - ERROR - Traceback (most recent call last):
  File "/opt/tboplayer/tboplayer.py", line 1103, in set_progress_bar_step
    self.progress_bar_var.set(int((self.omx.position * self.progress_bar_total_steps)/self.omx.timenf['duration']))
KeyError: 'duration'

Maksim-Bartoshyk, 2018-10-25:
Inspected debug and found that omx is called with -o property, but actual output is not set. I deleted ~/.tboplayer folder and this resolved the issue.
I remember that I played with audio output setting before the error occured and at some point I got all options for audio as grayed out and was not able to change them.

Unfortunately I don't remember actual steps to reproduce the issue but at some point I was able to get invalid configuration for audio, it caused running omx with invalid arguments and the error above.

Many thanks.

yutayu, 2018-10-27:
python /opt/tboplayer/tboplayer.py %F
2018-10-27 20:35:28,896 - /opt/tboplayer/debugging.pyc - DEBUG - started logging to file "/home/pi/.tboplayer/tboplayer.log"
2018-10-27 20:35:42,969 - /opt/tboplayer/debugging.pyc - DEBUG - >play track received
2018-10-27 20:35:42,971 - /opt/tboplayer/debugging.pyc - DEBUG - starting omxplayer with args: " -o hdmi  --vol 0"
2018-10-27 20:35:45,661 - /opt/tboplayer/debugging.pyc - DEBUG -             >Play: '/home/pi/Downloads/0/PIYO-007/PIYO-007A.mp4' with  -o hdmi  --vol 0
2018-10-27 20:35:45,662 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine: omx_starting
2018-10-27 20:35:45,663 - /opt/tboplayer/debugging.pyc - DEBUG -             <start play signal received from omx
2018-10-27 20:35:45,664 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine: omx_playing started
2018-10-27 20:35:46,103 - /opt/tboplayer/debugging.pyc - DEBUG -             >Send Command: p
2018-10-27 20:35:48,607 - /opt/tboplayer/debugging.pyc - DEBUG - Saving windowed geometry: "480x360+43+247"
2018-10-27 20:35:51,748 - /opt/tboplayer/debugging.pyc - DEBUG - >play track received
2018-10-27 20:35:51,749 - /opt/tboplayer/debugging.pyc - DEBUG -             >Send Command: p
2018-10-27 20:35:52,005 - /opt/tboplayer/debugging.pyc - DEBUG -             >Send Command: p
2018-10-27 20:36:03,955 - /opt/tboplayer/debugging.pyc - DEBUG - >play track received
2018-10-27 20:36:03,956 - /opt/tboplayer/debugging.pyc - DEBUG -             >Send Command: p
2018-10-27 20:36:04,007 - /opt/tboplayer/debugging.pyc - DEBUG -             >Send stop to omx
2018-10-27 20:36:04,160 - /opt/tboplayer/debugging.pyc - DEBUG -             <end play signal received
2018-10-27 20:36:04,160 - /opt/tboplayer/debugging.pyc - DEBUG -             <end detected at: 0.0
2018-10-27 20:36:04,161 - /opt/tboplayer/debugging.pyc - DEBUG - Saving windowed geometry: "480x360+43+247"
2018-10-27 20:36:04,566 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine: omx_ending
2018-10-27 20:36:04,567 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine : is omx process running -  False
2018-10-27 20:36:04,567 - /opt/tboplayer/debugging.pyc - DEBUG -             <omx process is dead
2018-10-27 20:36:04,979 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine: omx_closed
2018-10-27 20:36:04,980 - /opt/tboplayer/debugging.pyc - DEBUG - What next, single track so exit
Fatal Python error: PyEval_RestoreThread: NULL tstate

me, too

heniotierra, 2018-10-27:
@yutayu Are you also using omx options field? If yes, which options are you using? How to reproduce the issue?

heniotierra, 2018-10-27:
@Maksim-Bartoshyk What do you mean, the actual output is not set?

yutayu, 2018-10-27:
> @yutayu Are you also using omx options field? If yes, which options are you using? From the debug output, I see no special arguments. How to reproduce the issue?

no options when I use omxplayer command line.
and depends on video kind.

>is omx process running - False
is error I thought.

heniotierra, 2018-10-27:
> and depends on video kind.

You mean some file extensions don't cause this problem?

yutayu, 2018-10-27:
> > and depends on video kind.
> 
> You mean some file extensions don't cause this problem?

not extensions.
depends on file. some mp4 file occurs error not all mp4's

heniotierra, 2018-10-28:
@yutayu 
Please, send me the file that makes the program fail. I'll see what I can do when I have more time. My email is heniotster at gmail dot com.

yutayu, 2018-10-28:
@heniotierra no time line mp4.
sorry, This video is 1.5G. so huge.  and I couldn't  cut it , because no time line.

heniotierra, 2018-10-28:
Oh, ok, I will check those things. Thank you both!

heniotierra, 2018-10-30:
@Maksim-Bartoshyk Omxplayer has an option "both" for audio device, meaning analog and hdmi output at the same time, so I changed the code to replace the option "auto" to "both", since "auto" isn't even a device option, it merely forces either analog or hdmi. And "both" mode should work in every Raspberry Pi model with firmware from around 2014 or later. This also fixes your issue.

heniotierra, 2018-10-30:
@yutayu Now, your problem is another one, and it will take more time to be solved, because at the moment I don't have a PSU to start my RPI and test tboplayer with the file you sent me, since the PSU I used died a few months ago. :cry:

heniotierra, 2018-10-30:
@Maksim-Bartoshyk Can you confirm that the code in the branch [issue150](https://github.com/KenT2/tboplayer/tree/issue150) fixes you problem?

yutayu, 2018-10-30:
/home/pi/Documents/tinyproxy.conf

> @yutayu Now, your problem is another one, and it will take more time to be solved, because at the moment I don't have a PSU to start my RPI and test tboplayer with the file you sent me, since the PSU I used died a few months ago.

Never mind . 
I glad to see you noticed that errors. :)

Maksim-Bartoshyk, 2018-10-30:
> @Maksim-Bartoshyk Can you confirm that the code in the branch [issue150](https://github.com/KenT2/tboplayer/tree/issue150) fixes you problem?

Will try some time this week...

Maksim-Bartoshyk, 2018-10-31:
> > @Maksim-Bartoshyk Can you confirm that the code in the branch [issue150](https://github.com/KenT2/tboplayer/tree/issue150) fixes you problem?
> 
> Will try some time this week...

Works fine, thanks.

heniotierra, 2018-10-31:
No problem, you're welcome.

closed 2018-10-31 by heniotierra