KenT2/tboplayer #154: Playing  failed with Experimental OpenGL driver.

yutayu, 2018-11-17:
tboplayer
2018-11-17 11:43:26,578 - /opt/tboplayer/debugging.pyc - DEBUG - started logging to file "/home/pi/.tboplayer/tboplayer.log"
2018-11-17 11:44:13,656 - /opt/tboplayer/debugging.pyc - DEBUG - >play track received
2018-11-17 11:44:13,661 - /opt/tboplayer/debugging.pyc - DEBUG - starting omxplayer with args: " -o hdmi  --vol -3800"
2018-11-17 11:44:16,267 - /opt/tboplayer/debugging.pyc - DEBUG -             >Play: '/home/pi/Desktop/FKl1ZoCO9YXHEuFp.mp4' with  -o hdmi  --vol -3800
2018-11-17 11:44:16,269 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine: omx_starting
2018-11-17 11:44:16,269 - /opt/tboplayer/debugging.pyc - DEBUG -             <start play signal received from omx
2018-11-17 11:44:16,270 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine: omx_playing started
2018-11-17 11:44:19,921 - /opt/tboplayer/debugging.pyc - DEBUG - >stop received
2018-11-17 11:44:20,235 - /opt/tboplayer/debugging.pyc - DEBUG -       Service stop required signal
2018-11-17 11:44:20,236 - /opt/tboplayer/debugging.pyc - DEBUG -             >Send stop to omx
2018-11-17 11:44:20,452 - /opt/tboplayer/debugging.pyc - DEBUG -       [!] move_video failed
2018-11-17 11:44:20,453 - /opt/tboplayer/debugging.pyc - DEBUG - org.freedesktop.DBus.Error.ServiceUnknown: The name :1.23 was not provided by any .service files
2018-11-17 11:44:20,460 - /opt/tboplayer/debugging.pyc - DEBUG -       [!] move_video failed
2018-11-17 11:44:20,460 - /opt/tboplayer/debugging.pyc - DEBUG - org.freedesktop.DBus.Error.ServiceUnknown: The name :1.23 was not provided by any .service files
2018-11-17 11:44:20,801 - /opt/tboplayer/debugging.pyc - DEBUG -             <end play signal received
2018-11-17 11:44:20,802 - /opt/tboplayer/debugging.pyc - DEBUG -             <end detected at: 0.0
2018-11-17 11:44:20,803 - /opt/tboplayer/debugging.pyc - DEBUG - Saving windowed geometry: "793x461+384+119"
2018-11-17 11:44:21,198 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine: omx_ending
2018-11-17 11:44:21,199 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine : is omx process running -  False
2018-11-17 11:44:21,200 - /opt/tboplayer/debugging.pyc - DEBUG -             <omx process is dead
2018-11-17 11:44:21,588 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine: omx_closed
2018-11-17 11:44:21,589 - /opt/tboplayer/debugging.pyc - DEBUG - What next, break_required so exit
2018-11-17 11:44:32,916 - /opt/tboplayer/debugging.pyc - DEBUG - >play track received
2018-11-17 11:44:32,917 - /opt/tboplayer/debugging.pyc - DEBUG - starting omxplayer with args: " -o hdmi  --vol -3800 --win 384,119,1177,580 --aspect-mode letterbox --no-osd"
2018-11-17 11:44:33,602 - /opt/tboplayer/debugging.pyc - DEBUG -             >Play: '/home/pi/Desktop/FKl1ZoCO9YXHEuFp.mp4' with  -o hdmi  --vol -3800 --win 384,119,1177,580 --aspect-mode letterbox --no-osd
2018-11-17 11:44:33,602 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine: omx_starting
2018-11-17 11:44:33,603 - /opt/tboplayer/debugging.pyc - DEBUG -             <start play signal received from omx
2018-11-17 11:44:33,603 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine: omx_playing started
2018-11-17 11:44:38,075 - /opt/tboplayer/debugging.pyc - DEBUG - Saving windowed geometry: "793x461+46+50"
2018-11-17 11:44:45,682 - /opt/tboplayer/debugging.pyc - DEBUG - >stop received
2018-11-17 11:44:45,683 - /opt/tboplayer/debugging.pyc - DEBUG -       Service stop required signal
2018-11-17 11:44:45,684 - /opt/tboplayer/debugging.pyc - DEBUG -             >Send stop to omx
2018-11-17 11:44:46,209 - /opt/tboplayer/debugging.pyc - DEBUG - Saving windowed geometry: "793x461+41+50"
2018-11-17 11:44:46,633 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine: omx_ending
2018-11-17 11:44:46,634 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine : is omx process running -  False
2018-11-17 11:44:46,634 - /opt/tboplayer/debugging.pyc - DEBUG -             <omx process is dead
2018-11-17 11:44:47,041 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine: omx_closed
2018-11-17 11:44:47,042 - /opt/tboplayer/debugging.pyc - DEBUG - What next, break_required so exit
invalid command name "1926016176ytdl_update_messages_loop"
    while executing
"1926016176ytdl_update_messages_loop"
    ("after" script)
Fatal Python error: PyEval_RestoreThread: NULL tstate
/usr/local/bin/tboplayer: line 2:  5439 Aborted                 python /opt/tboplayer/tboplayer.py

with any videos . Thank you.

yutayu, 2018-11-17:
2018-10-28 21:38:43,254 - /opt/tboplayer/debugging.pyc - ERROR - Traceback (most recent call last):
  File "/opt/tboplayer/tboplayer.py", line 1074, in set_progress_bar
    self.progress_bar_step_rate = self.omx.timenf['duration']/self.progress_bar_total_steps
KeyError: 'duration'

2018-10-28 21:39:12,182 - /opt/tboplayer/debugging.pyc - ERROR - Traceback (most recent call last):
  File "/opt/tboplayer/tboplayer.py", line 1074, in set_progress_bar
    self.progress_bar_step_rate = self.omx.timenf['duration']/self.progress_bar_total_steps
KeyError: 'duration'

heniotierra, 2018-11-17:
Hi,

Does the video play, however? Seems like there's some kind of problem related the Dbus. Dbus is necessary to control omxplayer. Maybe you're executing two instances of OMXPlayer at the same time, sending "play" command twice too quickly? If you close tboplayer and then run from the terminal:

                sudo pkill omxplayer

And then, if you run tboplayer again, do the same happen?

yutayu, 2018-11-17:
> Hi,
>
> Does the video play, however? Seems like there's some kind of problem related the Dbus. Dbus is necessary to control omxplayer. Maybe you're executing two instances of OMXPlayer at the same [...]

I tried. but no output.

omxplayer /home/pi/Desktop/FKl1ZoCO9YXHEuFp.mp4
Video codec omx-h264 width 720 height 1280 profile 100 fps 30.000000
Audio codec aac channels 1 samplerate 44100 bitspersample 16
Subtitle count: 0, state: off, index: 1, delay: 0
V:PortSettingsChanged: 720x1280@30.00 interlace:0 deinterlace:0 anaglyph:0 par:1.00 display:0 layer:0 alpha:255 aspectMode:0
have a nice day ;)

yutayu, 2018-11-17:
Sound is coming , but Videos is not.

yutayu, 2018-11-17:
and This causes under experimental OpenGL driver from raspi-config. 
Can you try it?
Any video output fails and sound is good.

yutayu, 2018-11-17:
okok don't bother it.
omxplayer doesn't play it , too.
omxplayer problem also.

heniotierra, 2018-11-17:
ok

closed 2018-11-17 by heniotierra