KenT2/tboplayer #171: After omxplayer updating,Tboplayer got full screen and no click accepting.

yutayu, 2019-06-11:
pi@raspberrypi:~ $ apt show omxplayer
Package: omxplayer
Version: 20190607+gitf06235c-1
Status: install ok installed
Priority: optional
Section: video
Maintainer: Serge Schneider <serge@raspberrypi.org>
Installed-Size: 637 kB
Depends: libasound2 (>= 1.0.16), libavcodec57 (>= 7:3.2.14) | libavcodec-extra57 (>= 7:3.2.14), libavformat57 (>= 7:3.2.14), libavutil55 (>= 7:3.2.14), libc6 (>= 2.4), libdbus-1-3 (>= 1.9.14), libfreetype6 (>= 2.2.1), libgcc1 (>= 1:3.5), libpcre3, libstdc++6 (>= 5.2), libswresample2 (>= 7:3.2.14), zlib1g (>= 1:1.1.4), ca-certificates, dbus
Homepage: https://github.com/popcornmix/omxplayer
Download-Size: unknown
APT-Manual-Installed: yes
APT-Sources: /var/lib/dpkg/status
Description: Command line media player for Raspberry Pi
 .
 It plays all media files supported by the Raspberry Pi
 hardware. Supports subtitles.

heniotierra, 2019-06-11:
Hi @yutayu ,

You mean fullscreen works, but clicking over the screen while playing is unresponsive?

Regards,

yutayu, 2019-06-11:
@heniotierra :)
yes. and also no key response.
I didn't edit setting.

heniotierra, 2019-06-11:
Please, turn the debugger option on, restart the player and try that again. Then post the contents on the logs and the terminal. Logs are stored in .tboplayer folder in you user home.

yutayu, 2019-06-12:
[image]

pi@raspberrypi:~ $ tboplayer
2019-06-12 08:58:44,643 - /opt/tboplayer/debugging.pyc - DEBUG - started logging to file "/home/pi/.tboplayer/tboplayer.log"
2019-06-12 08:59:05,587 - /opt/tboplayer/debugging.pyc - DEBUG - >play track received
2019-06-12 08:59:05,591 - /opt/tboplayer/debugging.pyc - DEBUG - starting omxplayer with args: " -o hdmi  --vol -4000  "
2019-06-12 08:59:08,454 - /opt/tboplayer/debugging.pyc - DEBUG -             >Play: '/data2/SDAB-085.mp4' with  -o hdmi  --vol -4000  
2019-06-12 08:59:08,455 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine: omx_starting
2019-06-12 08:59:08,457 - /opt/tboplayer/debugging.pyc - DEBUG -             <start play signal received from omx
2019-06-12 08:59:08,458 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine: omx_playing started
2019-06-12 08:59:09,816 - /opt/tboplayer/debugging.pyc - DEBUG - >stop received
2019-06-12 08:59:10,175 - /opt/tboplayer/debugging.pyc - DEBUG -       Service stop required signal
2019-06-12 08:59:10,175 - /opt/tboplayer/debugging.pyc - DEBUG -             >Send stop to omx
2019-06-12 08:59:10,728 - /opt/tboplayer/debugging.pyc - DEBUG -             <end play signal received
2019-06-12 08:59:10,729 - /opt/tboplayer/debugging.pyc - DEBUG -             <end detected at: 0.0
2019-06-12 08:59:10,730 - /opt/tboplayer/debugging.pyc - DEBUG - Saving windowed geometry: "480x360+86+82"
2019-06-12 08:59:11,156 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine: omx_ending
2019-06-12 08:59:11,157 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine : is omx process running -  False
2019-06-12 08:59:11,157 - /opt/tboplayer/debugging.pyc - DEBUG -             <omx process is dead
2019-06-12 08:59:11,558 - /opt/tboplayer/debugging.pyc - DEBUG -       State machine: omx_closed
2019-06-12 08:59:11,559 - /opt/tboplayer/debugging.pyc - DEBUG - What next, break_required so exit
Fatal Python error: PyEval_RestoreThread: NULL tstate
/usr/local/bin/tboplayer: line 2: 21997 Aborted                 python /opt/tboplayer/tboplayer.py

yutayu, 2019-06-12:
Log file is there , but is empty.

yutayu, 2019-06-12:
I ran from terminal , full screen , esc key accept.
This is not original issue.

heniotierra, 2019-06-12:
This kind of problem is usually related to the dBus layer of the system, when tboplayer can't connect to omxplayer. Idk, I'll have to have a llok. But I'll have to find some free time for that, it's not happening so soon.  For now, if that works, maybe you can downgrade omxplayer.

yutayu, 2019-06-12:
@heniotierra I see.
I don't hurry :)

yutayu, 2019-08-05:
This issue is fixed by update on raspbian buster.

closed 2019-08-05 by yutayu