KenT2/tboplayer #175: when I using, some error come,would you please help

Kingwulin, 2020-02-10:
once I click the button stop or select the mp4 file,there is some error below ,also,the file can't play

2020-02-10 07:10:00,174 - /opt/tboplayer/lib/debugging.py - ERROR - Traceback (most recent call last):
  File "/opt/tboplayer/lib/debugging.py", line 58, in __call__
    return self.func()
TypeError: set() missing 2 required positional arguments: 'first' and 'last'

piningforfjords, 2020-03-18:
I'm seeing positional arguments as well.  I thought it might go away when I added some items to the list.  I notice the following error once I have some list elements.  These occur when an item is selected in the list:

2020-03-18 17:06:02,626 - /opt/tboplayer/lib/debugging.py - ERROR - Traceback (most recent call last):
  File "/opt/tboplayer/lib/debugging.py", line 57, in __call__
    return self.func(args)
  File "/opt/tboplayer/lib/tboplayer.py", line 1564, in select_track
    sel = event.widget.curselection()
AttributeError: 'tuple' object has no attribute 'widget'

2020-03-18 17:06:05,722 - /opt/tboplayer/lib/debugging.py - ERROR - Traceback (most recent call last):
  File "/opt/tboplayer/lib/debugging.py", line 57, in __call__
    return self.func(args)
  File "/opt/tboplayer/lib/tboplayer.py", line 1564, in select_track
    sel = event.widget.curselection()
AttributeError: 'tuple' object has no attribute 'widget'

2020-03-18 17:06:11,365 - /opt/tboplayer/lib/debugging.py - ERROR - Traceback (most recent call last):
  File "/opt/tboplayer/lib/debugging.py", line 57, in __call__
    return self.func(args)
  File "/opt/tboplayer/lib/tboplayer.py", line 1564, in select_track
    sel = event.widget.curselection()
AttributeError: 'tuple' object has no attribute 'widget'

heniotierra, 2020-03-18:
Sorry guys, didn't see your messages before! I will try to understand what is going on! Thank you!

heniotierra, 2020-03-18:
BTW, are you able to play the files?

piningforfjords, 2020-03-18:
No I wan't able to play the files.  I was able to play them on the command line directly with omxplayer.

piningforfjords, 2020-03-18:
In case it is useful I used the OMX menu to get details on one of the list items.  Here is the dialog that resulted.  Gave the file and path correct but other elements were empty.
[image]

heniotierra, 2020-03-19:
I think this is some problem with Tkinter, so I went ahead and opened a bug in cpython's Tkinter's bug tracker. 

https://bugs.python.org/issue40011?@ok_message=file%2048982%20created%0Ahgrepo%20387%20created%0Amsg%20364586%20created%0Aissue%2040011%20created&@template=item

heniotierra, 2020-03-19:
@piningforfjords @Kingwulin for now, I had to rollback the base branch to the point where Python 2 version was working. Hopefully, the issue described in the link above (in Python's issue tracker) will be fixed in cpython and I will be able to use Tkinter's Events. Either that, or the whole GUI will have to be rewritten, maybe using another GUI lib.

heniotierra, 2020-03-20:
The Python 3 version can be found here:
https://github.com/KenT2/tboplayer/tree/py3_version

heniotierra, 2020-03-21:
@Kingwulin @piningforfjords The error you two mentioned should be fixed by now. However, there might still be other issues. If you can test it, that would be great! I also fixed the setup script, hopefully. I will only merge the changes in the master branch after someone can confirm it works. As I mentioned, you can find the Python 3 version here:
https://github.com/KenT2/tboplayer/tree/py3_version

piningforfjords, 2020-03-21:
Cool will try it out tomorrow.  The python 2 version works well by the way.

heniotierra, 2020-03-21:
Good! 

You can use the commands below to install the Python 3 version through the setup script, but that will overwrite the Python 2 version currently installed in your system.

    cd ~ && wget https://github.com/KenT2/tboplayer/tarball/py3_version -O - | tar xz &&
    cd KenT2-tboplayer-* && chmod +x setup.sh && ./setup.sh

If you prefer avoiding overwritting the Python 2 version, you can copy-paste this URL in your browser's navigation bar to download the tarball, which you will then uncompress.

    https://github.com/KenT2/tboplayer/tarball/py3_version

And after uncompressing, just do in the terminal:

    cd tboplayer
    python3 .

piningforfjords, 2020-03-21:
Nah that is no way to test.  I burned a new Raspbian image.  Everything seems to run cleanly and all the functionality I checked worked well.  Thanks.  The setup script leaves the user in the /opt/tboplayer directory.  One suggestion is to kick them back to ~ when everything is installed.

heniotierra, 2020-03-22:
Good suggestion! Thank you too! :)

closed 2020-03-22 by heniotierra