KenT2/tboplayer #176: Setup requires some additional details

piningforfjords, 2020-03-18:
I found several things were needed to make the install work.  Without these steps the UI would never appear.  No errors showed themselves during installation (except the libav-tools below) or while trying to run the program.

alias python="python3"                          # change default to python3 
The following changes were made to the manual install
sudo pip3 install pexpect                           # pip3 instead of pip
sudo pip3 install requests
sudo apt-get install python3-magic          # python3 instead of python
sudo apt-get install python3-dbus            # not sure why this didn't catch on the first

libav-tools doesn't appear to have an install any longer part of raspbian now?

This gets the UI to launch but still unable to play.  The setup script was attempted but UI would not show up.  This was with the latest full 2020/02/13 Raspbian and update/upgrade

piningforfjords, 2020-03-18:
Here are the details when trying to install libav-tools:

Reading package lists... Done
Building dependency tree
Reading state information... Done
Package libav-tools is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  ffmpeg

heniotierra, 2020-03-20:
Actually, these problemas you mentioned are due to Python 2 being the main version of Python in your OS. In the setup I assumed you are using Python 3 as the base Python version.

piningforfjords, 2020-03-21:
Raspbian ships with python 2.7 as the default.  It is worth mentioning switching that over in the instructions or lots of people will get tripped up.

heniotierra, 2020-03-21:
You're right, of course. I did as you suggested, adding aliases for python and pip.

closed 2020-03-22 by heniotierra