KenT2/tboplayer #180: progress bar error / rasberry pi 4 / python 3

mikcaren, 2020-05-25:
Thanks for a great library.  I'm trying to run tboplayer on a rasberry pi 4 and received an "Error trying to set progress bar step" error. Here are the details:

pi@raspberrypi:~ $ python3 /opt/tboplayer/
>play track received
starting omxplayer with args: " -o local  --vol 0  "
            >Play: '/home/pi/Desktop/chris.mp4' with  -o local  --vol 0  
      State machine: omx_starting
            <start play signal received from omx
      State machine: omx_playing started
Error trying to set progress bar step
      State machine: omx_ending
      State machine : is omx process running -  False
            <omx process is dead
      State machine: omx_closed
What next, single track so exit

Per the troubleshooting guide - I am able to run and play the video with omxplayer via the command line.  I have also tried shutting down all instances of omxplayer and that didn't appear to solve the issue either.  I'm happy to run tests / give more info -- just let me know what is useful.  Thank you.

heniotierra, 2020-05-25:
Hi @mikcaren ,

Please, turn the debug option on, restart the player, reproduce the error, and then check what's in your debug.log file saved under /opt/tboplayer

mikcaren, 2020-05-25:
Thank you.  Yes - I was able to turn on the debug option, restart the player and reproduce the error.  Here is what was in the debug.log file in the /opt/tboplayer directory:

(14/05/2020 14:35:03): 'duration'
(14/05/2020 14:35:04): 'duration'

Thanks for the quick help and happy to do more tests etc.

heniotierra, 2020-05-25:
Ok, thank you for the report! I will check this out as soon as I can.

heniotierra, 2020-05-25:
Tell me, are you able to play other files? Or this happens with any files you try?

mikcaren, 2020-05-25:
I just tried it with a video I re-encoded with VLC and got the same error message.  I'm happy to try another file if you have any suggestions on types / formats. Also, I'm am launching the app from the terminal while I'm in the Desktop directory ... and noticed it's actually creating the current  debug.log file on the desktop.  Here are the contents:

(25/05/2020 14:32:44): 'duration'
(25/05/2020 14:32:45): 'duration'
(25/05/2020 15:50:30): 'duration'
(25/05/2020 15:50:30): 'duration'
(25/05/2020 16:41:46): 'duration'
(25/05/2020 16:41:47): 'duration'
(25/05/2020 16:42:42): 'duration'
(25/05/2020 16:42:43): 'duration'

I will try and find a couple different files to test.  Also, for the installation -- I had to install it via the "manual" instructions on the Github page (the automatic installer didn't work for my system).  It's a pretty fresh / new image of Rasbian.  Just letting you know in case it helps to diagnose the issue.  Thanks again.

heniotierra, 2020-05-25:
Alright! In the meantime, you can use the python 2 version, that can be found here:

    https://github.com/KenT2/tboplayer/tree/py2_version

It should be working fine.

mikcaren, 2020-05-26:
OK - thank you.  Yes -- will try the python 2 version.  I appreciate the suggestion.

I did a quick test on a .wav, .mp3 and .aiff files.  All had the same 'duration' issue and I was also able to play them with the command line omxplayer.

Thanks again for a great library and happy to test anytime.

mikcaren, 2020-05-26:
I was able to install the python 2 version with the installer and it works great!  Thank you!  

Let me know if there are any tests I can / should run with the python 3 version when the time is right.  Thanks again!

heniotierra, 2020-05-26:
Right, I will let you know. Glad I got the Python 2 version working. :)

heniotierra, 2020-05-26:
> Right, I will let you know. Glad you got the Python 2 version working. :)

closed 2020-10-26 by heniotierra