KenT2/tboplayer #182: No module named 'pexpect'

cripava, 2020-06-25:
hi there .

i followed the instruction on git . i tried to run it but i'm gting the following error

`Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/opt/tboplayer/__main__.py", line 67, in <module>
    from tboplayer import TBOPlayer
  File "/opt/tboplayer/lib/tboplayer.py", line 2, in <module>
    from ytdl import *
  File "/opt/tboplayer/lib/ytdl.py", line 2, in <module>
    import pexpect
ModuleNotFoundError: No module named 'pexpect'
`

cripava, 2020-06-25:
so i solved typing 

`sudo pip3 install pexpect`

maybe you want to know to add it on the dependencies

heniotierra, 2020-06-26:
pexpect is already in the dependencies. Maybe some error occurred during the installation and it wasn't installed.

closed 2020-10-26 by heniotierra