KenT2/tboplayer #183: set progress bar error problem

MichaelJWells, 2020-07-10:
I get the following message when I try to run a video in tboplayer: "**Error trying to set progress bar step**" and then omxplayer exits **without playing** the video.  

I have a new installation of Raspian Buster on a RPI 2 Rev B and I have followed all the manual install steps for the tboplayer which installed without error. I run the tboplayer from the command line with the following command: python3 /opt/tboplayer/  and when the window opens, I add a track and try to play it.  The error message appears in the background on the terminal CLI window I used to open the tboplayer.  [NB: I checked the debug box in the tboplayer options settings.]  

Testing shows that I am able to get the video to play normally with omxplayer from the command line. I also tested running the sudo pkill -9 omxplayer command in the command line as suggested in the installation page at github for tboplayer but this did not fix the error message.  I was able to run the video also using the omxplayerGUI application (for which I like less the GUI window layout generally in comparison to tboplayer).

Any suggestions about how to proceed to debug or fix this problem would be most appreciated.

heniotierra, 2020-07-10:
I don't have access to a Rapsberry Pi ATM, so I cannot fix this. My suggestion to try to circumvent it is making sure you're using python 3.7 or using the python 2 version. If you want to try to debug/fix it, I believe it might have something to do with the current time extraction in the method `_get_position` in lib/Omxplayer class. Maybe the position attribute is not being set correctly.

MichaelJWells, 2020-07-13:
> I don't have access to a Raspberry Pi ATM, so I cannot fix this for now. My suggestion to try to circumvent this is making sure you're using python 3.7 or using the python 2 version.

$ python3 --version
Python 3.7.3
$ python --version
Python 2.7.16

I installed the tboplayer for python2 and the tboplayer is now working though I imagine with fewer bells and whistles than the python3 version of tboplayer.

Since you do not have access to a RPI, I assume that development on this project is mostly at a standstill?

> If you want to try to debug/fix it, I believe it might have something to do with the current time extraction in the method _get_position in lib/OMXPlayer class: maybe the position attribute is not being set correctly.

Unfortunately, I have only a rudimentary level of knowledge of python programming and feel that the debugging task should be left up to those with more knowledge than I at this point in time. 

> And, unfortunately, the debug option is not very useful right now.

After installing the python2 version of the tboplayer, I happened to notice a debug.log file in the /home/pi directory instead of the previously indicated /opt/tboplayer/ directory. I do not know if this log file is related to the tboplayer version 2 or 3.

Thank you for your advice.

heniotierra, 2020-07-19:
> $ python3 --version
> Python 3.7.3
> $ python --version
> Python 2.7.16
>
> I installed the tboplayer for python2 and the tboplayer is now working though I imagine with fewer bells and whistles than the [...]

I expected to have access to my RPi in the next week.

closed 2020-10-26 by heniotierra