KenT2/tboplayer #185: Icon request

Botspot, 2020-09-21:
I am developing a RPi app store and I plan to include this software.
When users click "install", the app store will then clone your repository and run your recommended install commands.

One problem though: it needs a unique icon. Does tboplayer have an icon? If not then I'll have to leave it blank.

heniotierra, 2020-09-22:
That's nice! I suggest you add the python 2 version, since it's much more stable.
Well, I'm not a good designer, but I can make a simple icon for the program, sure!

Botspot, 2020-09-22:
Thank you! I myself have designed several good-looking logos, and if you'd like I can give feedback & offer suggestions.

Speaking of stability, a while back I had an issue searching youtube with tboplayer where the search results would never show up. Are you aware of that bug? If not, never mind and don't worry about it.

heniotierra, 2020-09-22:
Sure, I appreciate the help on the icon!

I am aware of this bug. The cause is that Youtube changed its search results page, so the script that was made to crawl it does not work anymore. I opened an issue reporting this problem today. This also happens in the python 2 version.

Botspot, 2020-09-22:
Great! You know what, I could design the whole icon for you if you want. Just give me a rough idea of what it should look like,

heniotierra, 2020-09-22:
That would be great! I thought of something really simple, maybe with a light blue font and transparent background, like this.

[image]

Botspot, 2020-09-22:
What does TBO stand for?

heniotierra, 2020-09-22:
Good question. Maybe @KenT2 can answer that. hehe

KenT2, 2020-09-22:
Thompson Baiter Omxplayer

Botspot, 2020-09-22:
I'm afraid your logo won't be suitable as an icon. It's a good start, but we have to remember it must be recognizable at small sizes.
This square below is the size of an icon.
[image]
And scaling yours down would make it hard to read:
[image]

Botspot, 2020-09-22:
OK, let me know what you think of these:
[image]
[image]

Botspot, 2020-09-22:
[image]

heniotierra, 2020-09-22:
Ooh, nice! I like the middle one better. 
[image]

BTW, you read my mind! I was thinking of something along these lines! :D

But I'd use the color scheme in here 

[image]

because using the inside of the circle in red seems like a reference to Youtube.

Botspot, 2020-09-22:
Well, it **is** a reference to YouTube. :) Because TBOplayer can play youtube.
I prefer the outer circle being black, as it comes across more as an O than otherwise.

If you don't like the reference to YouTube, we could change the color:
[image]

heniotierra, 2020-09-22:
Well, yes, it can play Youtube, but it can also play all of the most popular streaming services (last time I checked), that's why I thought of not making this reference. But I loved the icon! And I guess the reference doesn't hurt.

I like these two:
[image]

[image]

Ok, vote for the red one because it draws more attention. :)

Hey @KenT2 , would you have an opinion?

heniotierra, 2020-09-22:
> Well, yes, it can play Youtube, but it can also play all of the most popular streaming services (last time I checked), that's why I thought of not making this reference. But I loved the icon! And I [...]

Botspot, 2020-09-23:
@KenT2 @heniotierra Have we made a decision yet?

heniotierra, 2020-09-23:
I guess we can go with the red one. :)

Botspot, 2020-09-23:
OK. Attached is a zip file containing a bunch of sizes of your new icon.
Thank you.
[tbo.zip](https://github.com/KenT2/tboplayer/files/5272362/tbo.zip)

heniotierra, 2020-09-23:
Thank YOU for your work!

heniotierra, 2020-09-23:
@Botspot please, send an email if you need anyhting specific from me to make TBOPlayer installer work on the store you're building. It's heniotster at gmail dot com.

Botspot, 2020-10-03:
I sent you an email over a week ago. Have you seen it?
Basically, I'd like to see TBOPlayer ship with the new icon. (it would look a lot prettier than the python logo it has now)
And I also noticed that after installation, a mystery menu button appeared: Pure Data. What is it, and can `puredata` be uninstalled safely?

heniotierra, 2020-10-03:
Hi @Botspot ! Your email never arrived. 
Puredata? I don't see how that could be related to TBOPlayer. Try to open the shortcut with the text editor to see its contents. You can find such files under path `/usr/share/applications`.
About the icon, I should do that this weekend. I just haven't been having much free time as of late.

heniotierra, 2020-10-03:
Also, for transparency sake, I must say I'll dump the Python 3 version in favor of the python 2 one. The reason for that being that  OMXPlayer will be discontinued as can be seen in its github page, thus I don't think it's worth investing much in a Python 3 version. However, I should still fix and maintain the Python 2 version as long as OMXPlayer still works and has users. The efforts put into OMXPlayer will be put on developing VLC.

Botspot, 2020-10-03:
The Pure Data thing appeared in the menu at the same time I installed TBOPlayer. I would assume one of your APT dependencies managed to install it.

Looking forward to the icon change. :)

OK. Well I've already taken your suggestion and am using the Py2 version.

heniotierra, 2020-10-03:
Maybe it's some dependency doing that, yeah. Can you try to investigate that button? If it's showing in the menu, it's most probably present in the path I mentioned. By looking as its code through the text editor you should be able to know its source. You can paste the content here too, so I can take a look.

Botspot, 2020-10-03:
I can check it later, but it did go away after I ran:

    sudo apt purge puredata

Botspot, 2020-10-03:
Okay, I found it.
The rogue menu button is created at `/usr/share/applications/puredata.desktop`.
I uninstalled the `puredata` package, then ran the tboplayer install script. Sure enough, puredata was there again.

Botspot, 2020-10-03:
And this is the contents of `/usr/share/applications/puredata.desktop`: 
```
[Desktop Entry]
Type=Application
Name=Pure Data
GenericName=Pd
Comment=Visual dataflow programming platform for multimedia
Comment[ca]=Plataforma de programació visual per aplicacions multimèdia
Comment[de]=Grafische Datenflussprogrammierung für Multimedia
Comment[es]=Plataforma de programación visual para aplicaciones multimedia
Comment[fr]=Plateforme de programmation visuelle pour applications multimédia
Comment[it]=Piattaforma di programmazione visuale per applicazioni multimedia
Comment[pt]=Plataforma de programação visuais para multimedia
Icon=puredata.xpm
Exec=/usr/lib/puredata/tcl/pd-gui.tcl %F
Terminal=false
Categories=AudioVideo;Audio;
MimeType=text/x-puredata;text/x-maxmsp;
Keywords=Audio;Video;Programming;Synthesis;Analysis
```

heniotierra, 2020-10-03:
Ok, thanks. I'll look into that.

Botspot, 2020-10-03:
Yep - looks to be a dependency problem.
When I install TBO's dependencies manually beforehand, I get:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  gem gem-doc gem-extra gem-plugin-assimp gem-plugin-gmerlin gem-plugin-lqt
  gem-plugin-magick gem-plugin-v4l2 libassimp4 libftgl2 libgavl1 libglewmx1.13
  libgmerlin-avdec1 libmagick++-6.q16-8 libquicktime2 pd-libdir puredata
  puredata-core puredata-dev puredata-doc puredata-extra puredata-gui
  puredata-gui-l10n puredata-import puredata-utils
Suggested packages:
  pd-zexy v4l2loopback-dkms | v4l2loopback-modules pd-aubio pd-csound pd-pdp
  multimedia-puredata python-gobject-2-dbg python-gtk2-doc python-socks tix
  python-tk-dbg
The following NEW packages will be installed:
  gem gem-doc gem-extra gem-plugin-assimp gem-plugin-gmerlin gem-plugin-lqt
  gem-plugin-magick gem-plugin-v4l2 libassimp4 libftgl2 libgavl1 libglewmx1.13
  libgmerlin-avdec1 libmagick++-6.q16-8 libquicktime2 pd-libdir puredata
  puredata-core puredata-dev puredata-doc puredata-extra puredata-gui
  puredata-gui-l10n puredata-import puredata-utils python-gobject-2
  python-gtk2 python-requests python-tk tkdnd
```

Botspot, 2020-10-03:
BINGO! I'll let the apt output speak for itself:
```
pi@raspberrypi:~ $ sudo apt-get install tkdnd
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  gem gem-doc gem-extra gem-plugin-assimp gem-plugin-gmerlin gem-plugin-lqt
  gem-plugin-magick gem-plugin-v4l2 libassimp4 libftgl2 libgavl1 libglewmx1.13
  libgmerlin-avdec1 libmagick++-6.q16-8 libquicktime2 pd-libdir puredata
  puredata-core puredata-dev puredata-doc puredata-extra puredata-gui
  puredata-gui-l10n puredata-import puredata-utils
Suggested packages:
  pd-zexy v4l2loopback-dkms | v4l2loopback-modules pd-aubio pd-csound pd-pdp
  multimedia-puredata
The following NEW packages will be installed:
  gem gem-doc gem-extra gem-plugin-assimp gem-plugin-gmerlin gem-plugin-lqt
  gem-plugin-magick gem-plugin-v4l2 libassimp4 libftgl2 libgavl1 libglewmx1.13
  libgmerlin-avdec1 libmagick++-6.q16-8 libquicktime2 pd-libdir puredata
  puredata-core puredata-dev puredata-doc puredata-extra puredata-gui
  puredata-gui-l10n puredata-import puredata-utils tkdnd
0 upgraded, 26 newly installed, 0 to remove and 0 not upgraded.
Need to get 13.4 MB of archives.
After this operation, 32.1 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```
(I found that the package `tkdnd` is to blame, as it's the one that installs `puredata`)

Botspot, 2020-10-03:
An update: using `apt` with `--no-install-recommends` for `tkdnd` appears to work OK for TBOPlayer.

heniotierra, 2020-10-04:
That's nice @Botspot ! Thank you again for your help! :)

heniotierra, 2020-10-04:
@Botspot , will you need any icon size different 24x24? If not, I'll leave only the images with this size and another one for the readme.

heniotierra, 2020-10-04:
@Botspot The changes were merged. Next I'll fix the Youtube search and move the py2_version branch to master.

closed 2020-10-04 by heniotierra

Botspot, 2020-10-04:
> @Botspot , will you need any icon size different 24x24?
Not sure what you mean

heniotierra, 2020-10-04:
I meant for the app store. I left three icon sizes there: 24x24, 48x48, 128x128.

Botspot, 2020-10-05:
Oh don't worry. I have all the icon sizes I need.

Botspot, 2020-10-05:
BTW, here's my app store: https://github.com/Botspot/pi-apps

Botspot, 2020-10-05:
Hey, could you also add the icon to the running program's icon? (the icon that's displayed in the panel while TBOplayer is running)
Screenshot below:
[image]

heniotierra, 2020-10-05:
> BTW, here's my app store: https://github.com/Botspot/pi-apps
Cool, will have a look. Thank you for the link! :) 
> Hey, could you also add the icon to the running program's icon? (the icon that's displayed in the panel while TBOplayer is running)
Screenshot below:
Yep, it's in my to-do list.

heniotierra, 2020-10-05:
@Botspot There it is: Fixed the youtube search feature, and added the icon to the window. Tomorrow I should replace the master branch.

heniotierra, 2020-10-05:
Oh, and it's such a great idea that app of yours!

heniotierra, 2020-10-05:
I'll update this thread when I replace the master branch code.

Botspot, 2020-10-05:
It's not working for me.
```
pi@raspberrypi:~ $ python /opt/tboplayer/tboplayer.py
Traceback (most recent call last):
  File "/opt/tboplayer/tboplayer.py", line 75, in <module>
    from youtubesearchpython import SearchVideos
ImportError: No module named youtubesearchpython
```

heniotierra, 2020-10-05:
@Botspot DId you run the setup again? The new required lib can be installed with:

    pip install --user youtube-search-python

heniotierra, 2020-10-05:
@Botspot  there was a problem with this package installation in the setup script. That should be fixed now.0

Botspot, 2020-10-12:
Thank you one more time for all your help.
For future reference, I'm uploading the original vector TBOPlayer icon. It's in SVG format so you could turn it into any size icon you could ever want. (or change the colors if you had a SVG editor)
[tbo.svg.zip](https://github.com/KenT2/tboplayer/files/5366484/tbo.svg.zip)

heniotierra, 2020-10-25:
Hey @Botspot , I think tkinter does not allow SVG for window icons, at least the python2 tkinter.

Botspot, 2020-10-25:
> Hey @Botspot , I think tkinter does not allow SVG for window icons, at least the python2 tkinter. But thank you, that may come in handy in the future.

I understand that. But, a SVG comes in useful because SVGs aren't made up of pixels: instead they are vectors. So with a SVG image, you can *export* it to any size PNG you want, and the edges of the shapes will always stay clear.

Let's say 5 years down the road: you may want a 1024x1024 icon, but I am long gone. With a SVG, it's really easy: just opent he SVG in Inkscape or an equivalent, and you have your 1024x1024 png image.  
But if I never gave you that SVG, you would simply not be able to get a 1024x1024 image - or if you were, it would be badly pixelated.

Sending you the SVG ensures you can generate any icon size you want, without relying on me. Make sense?

heniotierra, 2020-10-26:
@Botspot  OFC, that's why I said it might come in handy in the future.
BTW, just to let you know, I replaced the master branch with the python 2 version as I said I would.