KenT2/tboplayer #188: Full Screen bug?

DRAGxMEx2xHELL, 2020-10-19:
HI

Appears sometimes when i play a YouTube video through TBOPlayer it will go full screen by default unless i play the video again, then it will pop up as windowed. Sometimes in fullscreen my mouse & keyboard would become unresponsive where id have to pull the power from the wall as a buttons, and combinations would not respond. Not sure if this is a issue, or i overlooked something.

heniotierra, 2020-10-19:
Hi,

Are you using the python 3 or pyhlthon 2 version?

DRAGxMEx2xHELL, 2020-10-19:
Sorry not sure, was using a raspberry pi 4 with TwisterOS 1.8.4

heniotierra, 2020-10-19:
@DRAGxMEx2xHELL Which version date is in the "about" window? BTW, thank you for reposting this.

DRAGxMEx2xHELL, 2020-10-19:
February 5th 2019 it appears

heniotierra, 2020-10-19:
Ok, I'll try and reproduce this.

DRAGxMEx2xHELL, 2020-10-20:
i ran into it 3 times over the course of like 2 hours.

DRAGxMEx2xHELL, 2020-10-21:
upon looking into it, the unresponsive mouse and keyboard as on my end, but other than that, it going full screen upon first video, seems i need to repeatedly hit esc for it to exit, unless i missed something.

heniotierra, 2020-10-21:
This only happens for streaming?

DRAGxMEx2xHELL, 2020-10-21:
Havent test anything else besides youtube streaming, sometimes get full screens where i cant exit, some i can, seems random. Is there any commands to exit full screen already in place?

heniotierra, 2020-10-21:
I think the standard keyboard commands from omxplayer should work just
fine. To quit, hit q.

heniotierra, 2020-10-23:
@DRAGxMEx2xHELL Sorry, I misread you question. To exit full screen, hit F11. If that does not work, it probably means an issue with dbus. This kind of problem generally happens when there are 2 instances of omxplayer running at the same time. To force OMXPlayer to open in windowed mode, you can also try to set the optional arguments (OMXplayer options) in the config screen. For example: --win 1,1,500,400

DRAGxMEx2xHELL, 2020-10-24:
Tried few things, and Q or F11 didnt let me exit full screen when it auto pops up at times.

heniotierra, 2020-10-25:
Ok, this PR fixes it: https://github.com/KenT2/tboplayer/pull/189
If you have any other problem, please, reach out! 
And thank you again, @DRAGxMEx2xHELL . :)

closed 2020-10-25 by heniotierra