KenT2/tboplayer #192: auto Fullscreen, minimize or litte window

fabianmendes, 2020-11-30:
Hi, excuse me I would like to play videos and audios (I am not tried yet downloading only audio) in the background while working but it gets into fullscreen automatically, also, when it does it, It is not intuitive to exit, and while it is fullscreen clicking anywhere is equivalent to clicking on whatever is behind the fullscreen instead of the fullscreen (It's like it is working as a screensaver --not bad if the user wants it instead of turn on the black screen, but it's really bad when it is not desired).

heniotierra, 2020-11-30:
Windowed mode does not work for you? It should word as default behavior.

fabianmendes, 2020-11-30:
It plays on fullscreen automatically. Let me try again anyways

fabianmendes, 2020-11-30:
Ok, I could disable de fullscreen at play, it's in options.. Otherwise, I'm still unavailable to minimize it and it keeps in front; the worst thing: I can't resize it at all. :L (by the way, this is an amazing app, don't misundertand me hehe))

heniotierra, 2020-12-01:
What happens when you try to resize the video? You should be able to resize by clicking and dragging the first mouse button over the button shown in the printscreen below. Maybe you can't see it because you have a small sceen. But you should see it if you drag the screen around.
[image]

>  don't misundertand me 
I'm glad you like it!

BTW, I just checked and yes, the default configuration is for windowed mode. Maybe you changed that somehow? Also, sometimes you may not get the video window layer, but AFAIK, this happens when you have another instance of omxplayer running in your system. If this happens, try to kill the omxplayer process.

fabianmendes, 2020-12-07:
I didn't say, excuse me. I could resize it! Thank you, but is there any way to hide it if it is a video...?

heniotierra, 2020-12-07:
> but is there any way to hide it if it is a video...

I never tried this, but it's possible to set a transparency in the video layer. We could have a button somewhere for this.