KenT2/tboplayer #193: Could you make omxplayer output to the selected ALSA output device?

Botspot, 2020-12-02:
Title says it all. I'd like if omxplayer obeyed my ALSA setup and outputted to the right audio device.

heniotierra, 2020-12-02:
I guess I can try. ;)

heniotierra, 2020-12-06:
@Botspot When do you suggest we do this? Upon a media playback? I'm thinking of doing that configuration upon first execution.

Botspot, 2020-12-06:
> @Botspot When do you suggest we do this? Upon starting a media playback? I'm thinking of doing that configuration upon first execution of tboplayer. We cannot change the output port while a media is being played.

What ever option is easiest I suppose. Doesn't omxplayer have a built-in ALSA output device flag?

heniotierra, 2020-12-06:
Yeah, I know, and tboplayer offers an ALSA output option. I not sure I
understand your request. My ideia is getting the current output port for
audio in the OS and using that for automatically setting OmxPlayer output.
I may have misunderstood your requst as I have very little knowlegde of
working with ALSA.

Botspot, 2020-12-06:
Basically, here's what happened:

- I have the Pi's analog output selected as the default audio output via ALSA.
- But TBOPlayer appears to ignore that output - it's always sent sound through my HDMI screen.

So I thought it might be a good idea if you were to change that, and make tboplayer obey the ALSA standard.

Botspot, 2020-12-06:
Okay sure enough, there's a setting for it. And it works fine.
[image]

So now I'm wondering: Why isn't ALSA output the default on a fresh tboplayer installation? (to me it seems like using alsa's selected output device would prove to satisfy most users better)

heniotierra, 2020-12-06:
> to me it seems like using alsa's selected output device would prove to satisfy most users bette0r

Maybe your're right. I've always used HDMI output.

heniotierra, 2020-12-06:
Done, it's in the master branch

closed 2020-12-06 by heniotierra