KenT2/tboplayer #196: Trying out on RPi 3

fabianmendes, 2020-12-11:
Hi, excuse me. I was running this on a Raspberry pi 3B+ And got this errors. I don't know.

It could be a minor issue (it's a trying out on raspberry pi 3, but it **should** work). But it's a bit interesting

```
~ $ python /opt/tboplayer/tboplayer.py
2020-12-11 10:41:15,430 - /opt/tboplayer/debugging.pyc - ERROR - Traceback (most recent call last):
  File "/opt/tboplayer/debugging.py", line 57, in __call__
    return apply(self.func, args)
  File "/opt/tboplayer/tboplayer.py", line 1746, in load_youtube_playlist
    self.go_ytdl(d.result,playlist=True)
  File "/opt/tboplayer/tboplayer.py", line 474, in go_ytdl
    self.ytdl.retrieve_youtube_playlist(url, youtube_media_format)
  File "/opt/tboplayer/ytdl.py", line 157, in retrieve_youtube_playlist
    process = pexpect.spawn(ytcmd, timeout=180, maxread=50000, searchwindowsize=50000)
  File "/usr/lib/python2.7/dist-packages/pexpect/pty_spawn.py", line 197, in __init__
    self._spawn(command, args, preexec_fn, dimensions)
  File "/usr/lib/python2.7/dist-packages/pexpect/pty_spawn.py", line 297, in _spawn
    cwd=self.cwd, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pexpect/pty_spawn.py", line 308, in _spawnpty
    return ptyprocess.PtyProcess.spawn(args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/ptyprocess/ptyprocess.py", line 321, in spawn
    raise exception
OSError: [Errno 8] Exec format error

```

[image]

heniotierra, 2020-12-11:
Hi! Maybe this is a problem in the debugger. You probably have debug option turned on. Do you also get an error when it's off? Which error?

fabianmendes, 2020-12-12:
Ok... I was replying to your message three times so far, this internet... Hahaha! Ok. I was saying:

May you please let me know _how can I do that_? Excuse me. And what do you mean when you say "an error **when it's off**? I think I've got none yet

heniotierra, 2020-12-12:
I mean if the error also occurs with the debugger off. You can find the option to turn off the debugger in Options > debug > save, and then restarting TBOPlayer.

Also, can you please send me your options file? It can be found in ~/.tboplayer/tboplayer.cfg

fabianmendes, 2020-12-12:
Sure, thanks. I will! Give me a half hour, please - thank you so much!

fabianmendes, 2020-12-13:
> I mean if the error also occurs with the debugger off. You can find the option to turn off the debugger in Options > debug, and then restarting TBOPlayer.
> 
> Also, can you please send me your options file? It can be found in `~/.tboplayer/tboplayer.cfg`.

It was already off:
[image]

Here is the option file:
```
[config]
audio = alsa
subtitles = off
mode = single
playlists =
tracks =
ltracks =
omx_options =
debug = off
youtube_media_format = mp4
omx_location = /usr/bin/omxplayer
ytdl_location = /usr/local/bin/youtube-dl
download_media_url_upon = add
youtube_video_quality = medium
geometry = 662x380+24+138
full_screen = 0
windowed_mode_coords = +200+200
windowed_mode_resolution = 480x360
forbid_windowed_mode = 0
cue_track_mode = 0
autoplay = 1
find_lyrics = 0
autolyrics_coords = +350+350
lang = en
subtitles_lang = en
ytdl_update = 1

```

after changing it to on, it gives me the same error. I've open the config file twice and it's ok. I mean, debug = _off_ and _on_ respectively.

After turning _on_ to try it out. it says:  (less lines than before, I guess)
```
2020-12-13 01:55:26,722 - /opt/tboplayer/debugging.pyc - DEBUG - started logging to file "/home/bridge/.tboplayer/tboplayer.log"
2020-12-13 02:01:08,064 - /opt/tboplayer/debugging.pyc - ERROR - Traceback (most recent call last):
  File "/opt/tboplayer/debugging.py", line 57, in __call__
    return apply(self.func, args)
  File "/opt/tboplayer/tboplayer.py", line 2195, in add_link
    self.add_url(self.video_link.get())
  File "/opt/tboplayer/tboplayer.py", line 1579, in add_url_from_search
    self.go_ytdl(link)
  File "/opt/tboplayer/tboplayer.py", line 472, in go_ytdl
    self.ytdl.retrieve_media_url(url, youtube_media_format)
  File "/opt/tboplayer/ytdl.py", line 150, in retrieve_media_url
    process = pexpect.spawn(ytcmd)
  File "/usr/lib/python2.7/dist-packages/pexpect/pty_spawn.py", line 197, in __init__
    self._spawn(command, args, preexec_fn, dimensions)
  File "/usr/lib/python2.7/dist-packages/pexpect/pty_spawn.py", line 297, in _spawn
    cwd=self.cwd, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pexpect/pty_spawn.py", line 308, in _spawnpty
    return ptyprocess.PtyProcess.spawn(args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/ptyprocess/ptyprocess.py", line 321, in spawn
    raise exception
OSError: [Errno 8] Exec format error

```

P.S: I've did it when I told you I will do this, but then I couldn't sit down and send it to you. Greetings

fabianmendes, 2020-12-18:
@heniotierra any idea?

heniotierra, 2020-12-18:
Hi Fabian,

I didn't have the chance to look at it. But I will do my best to get the
time soon.

fabianmendes, 2020-12-18:
Thaanks! No worries

https://github.com/

<https://github.com/KenT2/tboplayer/issues/196#issuecomment-748320806>, or
unsubscribe

heniotierra, 2020-12-19:
Hi @fabianmendes , can you describe how to reproduce the issue? Until now I wasn't able to.

fabianmendes, 2020-12-19:
I've just ran it on rpi3 (raspberry os), and tried to use it.. That's it

heniotierra, 2020-12-19:
The error you posted in the description of the issue apparently happened when trying to include a youtube playlist. But then you're saying it doesn'y work at all? I have a Raspberry Pi 3 B, and it works just fine for me.

fabianmendes, 2020-12-19:
No, i can open it but I cant play any music. Does yours add playlists?

heniotierra, 2020-12-20:
Yes, it works normally for me, playing files from the file system or
streams.

heniotierra, 2020-12-20:
@fabianmendes For the problem shown in the issue description, I'd suggest you to chcek that `youtube-d`l was installed correctly. Try to run in the terminal: 

    cat /usr/local/bin/youtube-dl

If you see nothing, that means youtube-dl isn't installed, possibly because an error occurred during installation, like youtube-dl's server's SSL certificate was out of date.

If you see `youtube-dl`'s binary content,then it's something else. My guess is youtube-dl cannot find the `python` environment variable, for some reason. I had this happen in Linux Mint, and I solved it by creating a symbolic link for `python3` as `python`. You can check if this is the case by doing in the terminal:

    python

If you get a "not found error", then this is related to python's path missing in the environment variables list.

fabianmendes, 2020-12-21:
All right thanks, ill do it this next morning

fabianmendes, 2020-12-21:
Same. But the problem was when adding a playlist and trying to add a track from the Youtube Search.  Also, after adding a track using URL, it just can't be played because of duration (I understand this like the track is not loaded yet, and it keeps like this: unplayable).
This is my _uname -a_
`Linux xyo-bridge 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux`
Here is the same log than comments below. Otherwise, do you think it may be because my flavor OS or the same thing than Mint? And thank you for your time and reply.
```
bridge@xyo-bridge:~ $ cat /usr/local/bin/youtube-dl
bridge@xyo-bridge:~ $ sudo apt-get install youtube-dl
[sudo] password for bridge: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
youtube-dl is already the newest version (2017.05.18.1-1).
0 upgraded, 0 newly installed, 0 to remove and 278 not upgraded.
bridge@xyo-bridge:~ $ 

```

heniotierra, 2020-12-22:
By the output from `cat /usr/local/bin/youtube-dl`, you do not have youtube-dl installed where it is expected to be. You can install it somewhere else and tell tboplayer where it is located though, thourgh a config option.

Anyway, I suggest you `apt remove` youtube-dl, because youtube-dl quickly gets out of date, which makes it eventually to stop working, and the version available in the OS repository is old and the maintainers take quite some time to update it. You should instal youtube-dll directly from its official page.

    sudo wget https://yt-dl.org/downloads/latest/youtube-dl -O /usr/local/bin/youtube-dl
    sudo chmod a+rx /usr/local/bin/youtube-dl

After this, see if it works for you.

heniotierra, 2021-01-26:
@fabianmendes Hi! If you managed to solve the problem, please close this issue, unless you think TBOPlayer still needs fixing. Thanks!

closed 2021-01-26 by heniotierra