KenT2/tboplayer #197: How does the youtube search work?

Botspot, 2020-12-12:
`youtube-dl` doesn't appear to have a youtube **search** feature.  
It appears you can give `youtube-dl` a keyword to search for, but then it will **download** the top X results!

That's not what I want.

I only want search **results**, containing the title of the video, and the URL. How does TBOPlayer's YT search function work? Does it use youtube-dl or something else?

Botspot, 2020-12-12:
Okay update: I've made tremendous progress with youtube-dl at https://github.com/Botspot/pi-apps/issues/116#issuecomment-743803001

Botspot, 2020-12-16:
It's about finished!  
[YouTubuddy](https://github.com/Botspot/youtubuddy) is one awesome tool, and it takes TBOPlayer's search function to the next level.
[image]

heniotierra, 2020-12-16:
Hi, @Botspot ! Now TBOPlayer uses a lib (youtube-search-python) for this, not youtube-dl. I didn't know youtube-dl had a search function. I guess I can have a look at it when I have the time. Thank you!

Botspot, 2020-12-16:
> Hi, @Botspot ! Now TBOPlayer uses a lib (youtube-search-python) for this, not youtube-dl. I didn't know youtube-dl had a search function. I guess I can have a look at it when I have the time. Thank you!

You're welcome for the info. Feel free to take a look at how YouTubuddy's bash script does it, and if you have any questions let me know.