KenT2/tboplayer #199: Open list doesn't download URLs

fabianmendes, 2021-04-16:
hi, excuse me. after opening a saved CSV list or playlist.. the app doesn't load them.

fabianmendes, 2021-05-05:
Ok, I've just seen that it only happens when the list of videos was added as Playlist. The issue is related to Adding Tracks without their respective individual URLs. I'll create a new issue.
[image]

closed 2021-05-05 by fabianmendes

heniotierra, 2021-05-06:
Hi Fabian.  You may find out that the sabe playlist fraturei is not very
useful for youtube videos cause the URLs saved are actually temporary. If
you try to access it some time later, It will fial tô load.

fabianmendes, 2021-05-06:
Hi @heniotierra thanks for the reply ! All right, thanks to know. I'll update #200 issue soon (I sadly wonder that) so I was thinking if we can add on the code something that could read videos ' URLs from playlist URL via **scrapping**. I'll try it out; the thing is that you use Python2.7 and I am not very well familiarized with it!

heniotierra, 2021-05-06:
No need for scrapping, that's unnecessary work. For this to work, we should just keep the original URL in memory, and upon saving the playlist, use that instead of the downloaded URL. Then, when loading the playlist, use the ytdl object to fetch the media URLs.

fabianmendes, 2021-05-06:
Ooh I get it, Instead of Saving a list of Tracks, we save the Playlistlist instead, right? but, How can we let know TBOPlayer python code when saving only the playlist URL and not tracks (because it will save tracks without HTTP URL on it)/