KenT2/tboplayer #200: Load Youtube Playlist doesn't add their respective URLs

fabianmendes, 2021-05-05:
And It can't be re-loaded as List, because, when you Save List, it saves them without any URL at all. You can see on #199 
The good news is that it's still possible to play them as Playlist. When you Save List, and Load list then, you can't play anything because there is not any URL! 

(Adding track by URL or Searching it with Youtube Browser, TBOPlayer Add the tracks' URL. When it is added as a YouTube playlist, it doesn't --but you can still play them. Kind of weird). Help, please. 'Cause I don't know how `go_ytdl` & `ytdl` actually work on code.
[image]

fabianmendes, 2021-06-24:
I want to help to improve Save List or getting  the tracks URLs fro YouTube playlist :$

heniotierra, 2021-06-25:
@fabianmendes Ok, I'll help you with some pointers.

heniotierra, 2021-06-25:
@fabianmendes Ok, I thought about this, and the easiest way to implement this feature would be by modifying the "Add youtube paylist" dialog (where you input the url) adding a "Save playlist to file" checkbox. If the user ticks the checkbox, the Dialog will enable a second text field labeled "File name" that would allow the user to enter a file name, and when the user click "Ok", if the File name has been provided, create a new file and put the url as contents of the file, then with that name provided by the user, specify the file extension as .csv and save it to the same folder as the normal playlist is being saved (it's a VARIABLE in the `tboplayer.options` object, please check that). Then, to open the file, create a new option in the "Playlist" to "Load Youtube playlist from file", use tk's "open file dialog" to get that file, than open it as a normal text file, and then just read the url and call `tboplayer.go_ytdl` as seen [here] passing the url to it.(https://github.com/KenT2/tboplayer/blob/d07f72c5b507f6b65f735162f5b784ed6ab520e8/tboplayer.py#L1746).

There are other alternatives to implment playlist saving for youtube-dl

heniotierra, 2021-06-25:
@fabianmendes 

In TBOPlayer's [_open_list](https://github.com/KenT2/tboplayer/blob/d07f72c5b507f6b65f735162f5b784ed6ab520e8/tboplayer.py#L1715
) method, you can see that it considers each line has two [columns](https://github.com/KenT2/tboplayer/blob/d07f72c5b507f6b65f735162f5b784ed6ab520e8/tboplayer.py#L1726)

fabianmendes, 2021-06-26:
Ok, I got a little lost, I've tried to follow the action of the functions and methods, I'll try again tomorrow.

Anyways, I didn't get very well how the playlist is read, I saw that a track with URL HTTP link reading is very similar. but, how does the program put the tracks from a playlist youtube loaded can turn on the list? I'm trying to aim at how tracks are listed in the TBOplayer and how are from a Youtube playlist link. There should be a way, also, I understand that the software stores the map or mp4 file temporarily or in streaming, right?

I'm thinking of scrapping again haha but only because I'm thinking now that the tracks youtube URLs are well saved or listed because we pasted their URLs while in Load Youtube Playlist we don't. And that's why you're suggesting to add to "Save [play]list" a script when it was from youtube playlist (because we would save the playlist's link itself to then being read as a playlist link too. Am I right?

I think scrapping may ain't be that bad but if you say we could do it better well let's do it. Also, I don't know if scrapping works on python 2 nor if Youtube let us scrap each track links from the playlist website page (I think yes it does, but I want to confirm all the above from this comment of mine first)

heniotierra, 2021-06-26:
FIrst of all, about my suggestion, I thought of an improvement:

Now, to your questions.

> If the user ticks the checkbox, this will enable a second text field labeled "File name" that would allow the user to enter a file name, and when the user click "Ok"

Instead of adding both the checkbox and a second text field, only add the checkbox, and after the user clicked "OK", if the user ticked the checkbox, then open tikinter's file dialog for saving the file (as seen [here](https://github.com/KenT2/tboplayer/blob/d07f72c5b507f6b65f735162f5b784ed6ab520e8/tboplayer.py#L1752)).

> how does the program put the tracks from a playlist youtube loaded can turn on the list?

We use a state machine to watch for the output of youtube-dl.

> There should be a way, also, I understand that the software stores the map or mp4 file temporarily or in streaming, right?

Youtube uses temporary url for the media stream (which is not the same as Youtube's site url for the video). Since they are temporary, if we save them to the file and try to restore the list later on, these link won't work.

> And that's why you're suggesting to add to "Save [play]list" a script when it was from youtube playlist (because we would save the playlist's link itself to then being read as a playlist link too. Am I right?

Yes.

For an alternative, I wouldn't do scraping. If you do on your teminal:

    youtube-dl --get-id https://www.youtube.com/watch?list=RDMMbmVZDopspPU

This will give you the full list of single video identifiers that belong in the list. So, create a routine to fetch this list and then save those to a file. When opening the playlist file (in a new menu item for "Load Youtube playlist from file"), use the identifiers and concatenate them with Youtube's url for watching videos, then call tboplayer.go_ytdl for each.

fabianmendes, 2021-07-02:
I was having so many troubles debugging it. Getting errors from dependencies (I installed all of them properly but finally, I got an error from them unfixable. What do you suggest to me? I use Pycharm editor.

Otherwise, as everything is in python 2.7, some dependencies are only for python 2.7; so is that why it can't be converted to Python 3 right?

heniotierra, 2021-07-02:
It can be converted to Python 3. The reason I decided to drop the Python 3 port is because it wasn't 100% working, and the development for omxplayer was discotinued, so if the software that's the base of TBOPlayer is now discotinued, there's not much reason to invest lots of efforts in a Python 3 port. Unless we modify TBOPlayer's dependency on omxplayer and make it use vlc or some other software.

fabianmendes, 2021-07-02:
Oh, Ok I get it, nice. Alright, got it!
 ~V. Fabian T· Mendes C.

heniotierra, 2021-07-02:
Which dependencies are you having problems with? And which errors? I don't think I can be much of a help if you added some new dependency that's causing errors.

fabianmendes, 2021-07-02:
urllibs, error from importing "Request"
 :L
 ~V. Fabian T· Mendes C.

heniotierra, 2021-07-02:
What does `sudo apt-get -y install python-requests` give you?

fabianmendes, 2021-07-02:
python-requests is on its latest version (2.21.0-1)